glikely/obs-ptz #1: VISCA over IP support

alessionossa, 2021-04-13:
I think **VISCA over IP** doesn’t need any explanation.

Any way we can help bring support for this essential feature?

glikely, 2021-04-17:
It is certainly on my to-do list, but I've not had a great amount of time to work on the plugin over the past several months.

I would certainly appreciate any help getting VISCA over IP enabled.

alessionossa, 2021-04-18:
Great! I'll see what I can do in the next few weeks.

If you can provide a little explanation on how the code is structured at the moment, it would make the work easier!

glikely, 2021-05-07:
I've just pushed an update that will make things easier by removing most of libvisca and replacing it with local code. The ViscaInterface class implements all the interaction with the camera. You could implement IP support by replacing/subclassing ViscaInterface with one that uses a network connection instead of QSerialPort

glikely, 2021-06-04:
Initial implementation of VISCA-over-IP is now in pushed to the visca-over-ip-test branch

commit 6743e17c (2021-06-04): Visca over IP WIP

Fixes: #1
commit 21d8fc22 (2021-06-05): Add VISCA over IP support

Adds the PTZViscaOverIP class implementing the protocol over IP. Most of
the protocol can be shared with the VISCA over Serial implementation.
PTZVisca is an abstract base class implementing the common bits.

Fixes: #1
closed 2021-06-05 by glikely