glikely/obs-ptz #7: It does not send the stop code all the time

maxdrfr, 2021-06-14:
I use this plugin on linux and thank you for this project.
There are some problems, when I press an arrow and release it, it does not always send the 03:03:ff.
The problem happens every time, I have to click on all the other arrows randomly so that it sends the standby code and the PTZ stops to move.

Do you have a solution ?

glikely, 2021-06-14:
It sounds like the plug-in is not receiving replys from the camera. Can you
attach the OBS log file to this issue?

glikely, 2021-06-14:
What happens if you click and hold the direction button for  2s?

maxdrfr, 2021-06-14:
If I hold the arrow for more 3-4 sec, the stop code is sent but randomly.
If I hold the arrow less than 1 sec, the stop is never sent.

Log with the issue:

```
00:44:30.235: VISCA UDP --> 02:00:00:01:00:00:00:00:01
00:44:30.235: VISCA UDP --> 01:00:00:05:00:00:00:01:81:01:00:01:ff
00:44:31.285: PTZVisca::timeout() 0x7fc9f400d770
00:44:45.447: VISCA UDP --> 02:00:00:01:00:00:00:00:01
00:44:45.447: VISCA UDP --> 01:00:00:05:00:00:00:01:81:01:00:01:ff
00:44:46.408: PTZVisca::timeout() 0x7fc9f400d770
00:44:47.373: VISCA UDP --> 01:00:00:09:00:00:00:02:81:01:06:01:00:00:03:03:ff
00:44:48.408: PTZVisca::timeout() 0x7fc9f400d770
00:44:48.533: VISCA UDP --> 01:00:00:09:00:00:00:03:81:01:06:01:0a:00:02:03:ff
00:44:49.484: PTZVisca::timeout() 0x7fc9f400d770
00:44:51.229: VISCA UDP --> 01:00:00:09:00:00:00:04:81:01:06:01:0a:00:02:03:ff
00:44:52.279: PTZVisca::timeout() 0x7fc9f400d770
00:44:52.735: VISCA UDP --> 01:00:00:05:00:00:00:05:81:01:06:04:ff
00:44:53.785: PTZVisca::timeout() 0x7fc9f400d770
00:45:07.693: VISCA UDP --> 01:00:00:09:00:00:00:06:81:01:06:01:0a:00:02:03:ff
00:45:08.744: PTZVisca::timeout() 0x7fc9f400d770
00:45:10.943: VISCA UDP --> 01:00:00:09:00:00:00:07:81:01:06:01:0a:00:02:03:ff
00:45:11.908: PTZVisca::timeout() 0x7fc9f400d770
00:45:16.229: VISCA UDP --> 01:00:00:09:00:00:00:08:81:01:06:01:0a:00:01:03:ff
00:45:17.279: PTZVisca::timeout() 0x7fc9f400d770
00:45:20.677: VISCA UDP --> 01:00:00:09:00:00:00:09:81:01:06:01:0a:00:02:03:ff
00:45:21.727: PTZVisca::timeout() 0x7fc9f400d770
00:45:24.261: VISCA UDP --> 01:00:00:09:00:00:00:0a:81:01:06:01:0a:00:01:03:ff
00:45:25.311: PTZVisca::timeout() 0x7fc9f400d770
00:45:27.973: VISCA UDP --> 01:00:00:09:00:00:00:0b:81:01:06:01:0a:00:01:03:ff
00:45:28.923: PTZVisca::timeout() 0x7fc9f400d770
00:45:29.125: VISCA UDP --> 01:00:00:09:00:00:00:0c:81:01:06:01:0a:00:01:03:ff
00:45:30.075: PTZVisca::timeout() 0x7fc9f400d770
00:45:30.165: VISCA UDP --> 01:00:00:09:00:00:00:0d:81:01:06:01:0a:00:02:03:ff
00:45:31.215: PTZVisca::timeout() 0x7fc9f400d770
00:45:31.261: VISCA UDP --> 01:00:00:09:00:00:00:0e:81:01:06:01:0a:00:01:03:ff
00:45:32.311: PTZVisca::timeout() 0x7fc9f400d770
00:45:33.044: VISCA UDP --> 01:00:00:09:00:00:00:0f:81:01:06:01:0a:00:02:03:ff
00:45:33.995: PTZVisca::timeout() 0x7fc9f400d770
00:45:35.381: VISCA UDP --> 01:00:00:09:00:00:00:10:81:01:06:01:0a:00:01:03:ff
00:45:36.408: PTZVisca::timeout() 0x7fc9f400d770
00:45:36.557: VISCA UDP --> 01:00:00:09:00:00:00:11:81:01:06:01:0a:00:02:03:ff
00:45:37.507: PTZVisca::timeout() 0x7fc9f400d770
00:45:39.469: VISCA UDP --> 01:00:00:09:00:00:00:12:81:01:06:01:0a:00:01:03:ff
00:45:40.419: PTZVisca::timeout() 0x7fc9f400d770
00:45:40.478: VISCA UDP --> 01:00:00:05:00:00:00:13:81:01:06:04:ff
00:45:41.428: PTZVisca::timeout() 0x7fc9f400d770
```

glikely, 2021-06-14:
That's what I thought. The plug-in is not receiving the reply packets from the camera. Holding the button down allows the plug-in to timeout, which clears the command queue, before the stop command is sent.

The missing replys could be a firewall issue, it could be a network configuration problem, or the camera isn't sending replies. What kind of camera are you using?

Regardless, I think the right thing to do is send commands packets unconditionally instead of waiting for acknowledgment packets. That will give lower latency on camera control command, and it will eliminate the problem you're seeing. Waiting for acknowledgements is really only needed for inquiry packets.

maxdrfr, 2021-06-14:
Thats right, it's not a network or firewall issue, the device just listen without aknowledgment.
Before your plugin, I was using https://github.com/jvkolyadich/VISCA-IP without problems and I doesn't know that viscaip have to reply to each packets ;)

closed 2021-06-16 by glikely

glikely, 2021-06-16:
Try this test build:
https://github.com/glikely/obs-ptz/actions/runs/944188476

maxdrfr, 2021-06-16:
Thanks for the fix, sorry I'm using Linux.
But I'm gonna build the last commits soon.

maxdrfr, 2021-06-16:
It works great thank you.

But now I have to unplug the game controller (PS3 controller) because there is a conflict and the plugin sends a series of commands in a row and it disturbs the camera with the "Enable gamepad" box checked or not.

info: VISCA UDP --> 01:00:00:09:00:00:00:03:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:04:81:01:06:01:00:00:03:03:ff
Attempted path: share/obs/obs-studio/images/overflow.png
Attempted path: /usr/share/obs/obs-studio/images/overflow.png
info: VISCA UDP --> 01:00:00:09:00:00:00:05:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:06:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:07:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:08:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:09:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0a:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0b:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0c:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0d:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0e:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:0f:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:10:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:11:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:12:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:13:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:14:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:15:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:16:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:17:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:18:81:01:06:01:00:00:03:03:ff
info: VISCA PTZ timeout
info: VISCA UDP --> 01:00:00:09:00:00:00:19:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1a:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1b:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1c:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1d:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1e:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:1f:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:06:00:00:00:20:81:01:04:07:00:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:21:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:22:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:23:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:24:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:25:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:26:81:01:06:01:00:00:03:03:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:27:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:28:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:29:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2a:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2b:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2c:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2d:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2e:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:2f:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:30:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:31:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:32:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:33:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:34:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:35:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:36:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:37:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:38:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:39:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3a:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3b:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3c:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3d:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3e:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:3f:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:40:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:41:81:01:06:01:00:01:03:02:ff
[... 135 lines cut ...]
info: VISCA UDP --> 01:00:00:09:00:00:00:c9:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:ca:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:cb:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:cc:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:cd:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:ce:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:cf:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:d0:81:01:06:01:00:01:03:02:ff
[... cut ...]
info: VISCA UDP --> 01:00:00:09:00:00:00:db:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:dc:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:dd:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:de:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:df:81:01:06:01:00:01:03:02:ff
info: VISCA UDP --> 01:00:00:09:00:00:00:e0:81:01:06:01:00:01:03:02:ff
info: ==== Shutting down ==================================================

glikely, 2021-06-16:
The gamepad support is dodgy and needs to be rewritten. Please file another issue for tracking it

Mazecreator, 2021-06-18:
I had the same problem, your update on 6/16 fixed the problem for me as well.

commit 2bc30d63 (2021-06-16): VISCA: send commands immediately without queueing

VISCA commands, unlike enquiries, can be sent immediately without
waiting for acknowledgement because the return value doesn't need to be
matched to the specific outgoing command. Change send function to send
commands immediately, but still queue inquiries. This will make the
controls more responsive because there isn't a time lag to sending the
command to the camera.

This also fixes the problem of not sending the stop command if the
camera has not responded. Now the commands get sent unconditionally,
even if the camera is not attached.

Fixe [... cut]