glikely/obs-ptz #14: plugin not loading in (distro or latest) OBS on Ubuntu linux

braverock, 2021-07-04:
After building the shared object file:

[build.log](https://github.com/glikely/obs-ptz/files/6760141/build.log)

I've tried placing it everywhere suggested by various OBS forum posts:

```
/usr/lib/obs-plugins

/usr/share/obs/obs-plugins/

$HOME/.config/obs-studio/plugins/
```

but OBS does not load the plugin.  I've checked that file permissions on the .so object are the same as other libraries that OBS loads, I've run strace to see if it looks in those directories (it does), 

[strace.log](https://github.com/glikely/obs-ptz/files/6760131/strace.log)

I've double checked that I have all build dependencies.  I updated OBS from the version that is in the Ubuntu repositories to the official stable OBS repository.

I'm just stumped.

[2021-07-03 07-41-15.txt](https://github.com/glikely/obs-ptz/files/6760133/2021-07-03.07-41-15.txt)

braverock, 2021-07-04:
file:///home/brian/mule_dev/feedback/ptz/obs-ptz/build/CMakeCache.txt

cmake cache file has more paths, etc.

glikely, 2021-07-06:
I have no idea. The Linux build works for me on Debian, but I don't have access to my Linux box at the moment. I'll investigate when I do.

I'm simply symlinking the .so file into the /usr/lib/obs-plugins directory which is all I needed to do.

closed 2021-09-16 by glikely