glikely/obs-ptz #16: Feature request: ONVIF protocol support

glikely, 2021-07-08:
ONVIF is used by several makes of IP CCTV cameras

Mazecreator, 2021-07-30:
ONVIF would be powerful, I would love to see this added, I started to look up ways to do it and found this library:
https://github.com/Tereius/libONVIF

There is a reference guide as well:
https://www.onvif.org/wp-content/uploads/2016/12/ONVIF_WG-APG-Application_Programmers_Guide-1.pdf

ldfranca, 2021-08-02:
It would be the best update for the Plugin, I assemble transmission systems and use many PTZ cameras with ONVIF protocol and to control them always depending on an external system. With this ONVIF option in the plugin would only use OBS.

Nafru, 2022-01-06:
I am using a GenBolt PTZ camera that runs ONVIF.

I have found that there are several GitHub pulls and libraries already out for ONVIF. 
I am not sure how they bolt into your software.

https://github.com/webermarci/onvif-pull

https://github.com/Tereius/libONVIF

https://github.com/RichardoMrMu/gsoap-onvif

https://github.com/quatanium/python-onvif

dslat, 2022-02-24:
I am using Hikvision IP PTZ cameras with OBS Studio at several sites to stream video but currently have to use a separate web browser window to control the camera ptz and presets. Integrated ONVIF control would be wonderfully useful. 
I recently used a FoMaKo NDI PTZ IP camera using VISCA and the OBS-PTZ plugin worked great.

Could the C++ code for ONVIF IP Camera control in the following library be of help getting it up and running:

https://github.com/rocky-Cplusplus/onvif

michaeleino, 2022-04-19:
This code is working fine on zoneminder with hikvision
https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/HikVision.pm

May it help for hikvision cams ?

I can help testing for this feature, is there any expectation for coming up?
It is a really added value!
Thanks 😊

glikely, 2023-02-22:
ONVIF support is now enabled in mainline. The next release of obs-ptz (v0.16.0) will have it included. I'll tag a test release today. Please give it a try and report back.

closed 2023-02-22 by glikely

jonata, 2023-02-24:
Excelent, I tested here on 3 cameras I have access, Pan and Tilt are working fine (the devices don't have optical zoom). The presets are not working, though. I will investigate if this is something in my end on something missing on the code. If so I open an issue or PR. Thank you so much for the test release!

glikely, 2023-02-24:
> Excelent, I tested here on 3 cameras I have access, Pan and Tilt are working fine (the devices don't have optical zoom).

Thanks for testing.

> The presets are not working, though. I will investigate if this is something in my end on something missing on the code. If so I open an issue or PR. Thank you so much for the test release!

Presets are broken for ONVIF. I spent a bit of time trying to get it working, but I've decided instead to overhaul the presets backend to be able to handle ONVIF presets with string keys. The current code is hard coded to use integer preset numbers which don't map nicely. It will take some time to get this implemented.