glikely/obs-ptz #19: Cisco Precision HD camera and OBS PTZ compatibility

MarkoS65, 2021-07-09:
There are some difference in commands that Cisco implemented so Sony VISCA don't works with Cisco cameras (see attached user guide, from page 28). When using PTZ plugin in OBS, only pan and tilt commands works (not Up+Right... just one at a time).
Working:
-pan left
-pan right
-tilt down
-tilt upp
Working different:
-pan left+ tilt up: goes up and to the left to limit position.
-pan left+ tilt down: goes down and to the left to limit position.
-pan right+ tilt up: goes up and to the right to limit position.
-pan right+ tilt down: goes down and to the right to limit position.
-speed slider: when in bottom position, no movements, every other position, max speed.
Don't working:
-home button
-any preset function

Commands Cisco uses according user manual (attached):
Command; Command Packet; Comments
Zoom_Stop; 8x 01 04 07 00 ff ;
Zoom_Tele; 8x 01 04 07 2p ff ; p = speed parameter,
Zoom_Wide; 8x 01 04 07 3p ff ; a (low) to b (high)
Zoom_Direct; 8x 01 04 47 0p 0q 0r 0s ff ; pqrs: zoom position
PT_Stop;  8x 01 06 01 03 03 03 03 ff
PT_Reset; 8x 01 06 05 ff ; Reset pan/tilt to center position. This also re–synchronizes the motors.
PT_Up; 8x 01 06 01 0p 0t 03 01 ff  ; p: pan speed , t: tilt speed
PT_Down; 8x 01 06 01 0p 0t 03 02 ff
PT_Left; 8x 01 06 01 0p 0t 01 03 ff
PT_Right; 8x 01 06 01 0p 0t 02 03 ff
PT_UpLeft; 8x 01 06 01 0p 0t 01 01 ff
PT_UpRight; 8x 01 06 01 0p 0t 02 01 ff
PT_DownLeft; 8x 01 06 01 0p 0t 01 02 ff
PT_DownRight; 8x 01 06 01 0p 0t 02 02 ff
PT_Direct; 8x 01 06 02 0p 0t 0q 0r 0s 0u 0v 0w 0x 0y ff ; p: max pan speed,  t: max tilt speed, qrsu: pan position, vwxy: tilt position
PTZF_Direct; 8x 01 06 20 0p 0q 0r 0s 0t 0u 0v 0w 0x 0y 0z 0g 0h 0i 0j 0k ff ; Sets all motors in one operation.
pqrs: pan, tuvw: tilt, xyzg: zoom, hijk: focus, Attempts to linearize movement for pan and tilt. The focus value will not be used if the camera is in continuous autofocus mode. NOTE: Never route this message through Sony cameras

[precisionhd_1080p-720p_camera_user_guide.pdf](https://github.com/glikely/obs-ptz/files/6793066/precisionhd_1080p-720p_camera_user_guide.pdf)

glikely, 2021-07-09:
Can you attach the OBS log file from running the plugin with the Cisco camera? Would be ideal if you can annotate with what command was run at each point in the log.

wasn-eu, 2021-07-12:
would like to add that the zoom buttons do not work.

wasn-eu, 2021-07-12:
I have a Cisco PHD 1080p12x connected over serial with the same issues as MarkoS65.
I see this in my logs:

`With Speed Slider at about 10%
21:50:18.116: VISCA --> 81:01:04:07:03:ff	<--- Press Zoom Wide
21:50:18.216: VISCA --> 81:01:04:07:00:ff	<--- Release Zoom Wide

21:51:11.264: VISCA --> 81:01:04:07:02:ff	<--- Press Zoom Tele
21:51:12.174: VISCA --> 81:01:04:07:00:ff	<--- Release Zoom Tele

21:51:56.218: VISCA --> 81:01:06:04:ff		<--- Press Home

21:52:45.391: VISCA --> 81:01:04:3f:01:00:ff	<--- Save Preset 1

21:53:46.654: VISCA --> 81:01:04:3f:02:00:ff	<--- Recall Preset 1

With Speed Slider at 0%
21:55:04.673: VISCA --> 81:01:06:03:14:14:0f:0f:0f:0e:00:00:00:00:ff	<--- Press Left
21:55:04.826: VISCA --> 81:01:06:01:00:00:03:03:ff			<--- Release Left

With Speed Slider at 100%
21:55:44.436: VISCA --> 81:01:06:01:14:00:01:03:ff			<--- Press Left
21:55:44.632: VISCA --> 81:01:06:01:00:00:03:03:ff			<--- Release Left

`

glikely, 2021-07-12:
Can you attach the entire log file please? I'd like to see the traffic coming back from the camera too.

wasn-eu, 2021-07-12:
here is the full log, with pushing buttons in this order:

Speed Fader at 100%, left, home, wide, tele, Save preset 1, recall preset 1
Speed fader at about 0%, left, home, wide, tele
Speed fader at about 50%, left, home, wide, tele

23:02:27.129: VISCA --> 81:01:06:01:14:00:01:03:ff
23:02:27.211: VISCA --> 81:01:06:01:00:00:03:03:ff
23:02:27.927: VISCA --> 81:01:06:04:ff
23:02:28.970: VISCA --> 81:01:04:07:03:ff
23:02:29.077: VISCA --> 81:01:04:07:00:ff
23:02:31.285: VISCA --> 81:01:04:07:02:ff
23:02:31.411: VISCA --> 81:01:04:07:00:ff
23:02:32.494: VISCA --> 81:01:06:01:00:00:03:03:ff
23:02:32.494: VISCA --> 81:01:04:07:00:ff
23:02:35.828: VISCA --> 81:01:04:3f:01:00:ff
23:02:39.127: VISCA --> 81:01:04:3f:02:00:ff
23:02:42.101: VISCA --> 81:01:06:03:14:14:0f:0f:0f:0e:00:00:00:00:ff
23:02:42.241: VISCA --> 81:01:06:01:00:00:03:03:ff
23:02:43.161: VISCA --> 81:01:06:04:ff
23:02:43.922: VISCA --> 81:01:04:07:03:ff
23:02:44.045: VISCA --> 81:01:04:07:00:ff
23:02:44.939: VISCA --> 81:01:04:07:02:ff
23:02:45.077: VISCA --> 81:01:04:07:00:ff
23:04:41.944: VISCA --> 81:01:06:01:0a:00:01:03:ff
23:04:42.517: VISCA --> 81:01:06:01:00:00:03:03:ff
23:04:43.695: VISCA --> 81:01:06:04:ff
23:04:44.636: VISCA --> 81:01:04:07:03:ff
23:04:45.297: VISCA --> 81:01:04:07:00:ff
23:04:46.011: VISCA --> 81:01:04:07:02:ff
23:04:46.705: VISCA --> 81:01:04:07:00:ff

the camera does not respond to zoom commands
when speed fader is 0% there is no movement
when speed fader is above 0% the cam moves always at 100%

MarkoS65, 2021-07-12:
Hi.

I made test and the log file is attached. For some reason there were error messages about JPEG missing, 25 per second so it took a while to clean upp the file.
After last preset trial, i had speed slider at 0%, 10%, 20% and 100%. As for WASN above, at 0% no moves and at any other position the speed was the same.
I tried to compare commands in ptz-visca.cpp file, but couldn't see how the speed is implemented in the string. Commands seems the same. Maybe, for HOME button the RESET command should be used.

Best regards
Marko.S

[2021-07-12 20-09-36.txt](https://github.com/glikely/obs-ptz/files/6804465/2021-07-12.20-09-36.txt)

glikely, 2021-07-12:
I’ve just released v0.5.0. Can you do the same thing and post the log file with the new version? I don’t think it will fix anything, but it will give me more information about the camera that will help debugging.

glikely, 2021-07-13:
v0.5.0 may fix the zoom problem. In the last release I switched the zoom command to the one that includes the speed argument.

wasn-eu, 2021-07-13:
zoom is working now

glikely, 2021-07-13:
@wasn-eu thanks for checking. Are diagonal moves still broken?

MarkoS65, 2021-07-13:
Hi Tested this evening.

Zoom is working
Speed regulation for zoom is working.

The rest is same as before.

Logg file attached.

BRG
Marko.S
[2021-07-13 17-41-13.txt](https://github.com/glikely/obs-ptz/files/6811516/2021-07-13.17-41-13.txt)

wasn-eu, 2021-07-13:
diagonal moves are working fine.

glikely, 2021-07-13:
@wasn-eu , I don't see any reply messages from the camera in your log. Is the serial receive line wired up?

glikely, 2021-07-13:
@MarkoS65 , everything in the log looks right to me for the pan tilt commands. A speed value is getting encoded in the correct byte. I don't know what the problem could be there.

For the diagonal, I have one idea that I might try tomorrow

glikely, 2021-07-14:
I've just applied to hacks to the ci-test branch to try out some ideas. Please give this build a spin and post the log files. The two commits should get rid of the VISCA error message in the log and *might* solve the diagonal move problem that @MarkoS65 is having.

https://github.com/glikely/obs-ptz/actions/runs/1029649080

wasn-eu, 2021-07-14:
just played a little with the fifo and buffer settings of the com port.
now i receive packet from the camera.
But mwith many error:

```
18:47:33.385: ViscaUART::receive_datagram():482: VISCA <-- 88:30:02:ff
18:47:33.386: VISCA Interface COM4: 1 camera found
18:47:33.386: ViscaUART::send():476: VISCA --> 88:01:00:01:ff
18:47:33.386: ViscaUART::send():476: VISCA --> 81:09:00:02:ff
18:47:33.403: ViscaUART::receive_datagram():482: VISCA <-- 90:60:41:ff
18:47:33.403: PTZVisca::receive():369: VISCA HDMI received error: 90:60:41:ff
18:47:33.403: ViscaUART::send():476: VISCA --> 81:09:00:02:ff
18:47:33.415: ViscaUART::receive_datagram():482: VISCA <-- 90:60:41:ff
18:47:33.415: PTZVisca::receive():369: VISCA HDMI received error: 90:60:41:ff
```

attached the full logs

glikely, 2021-07-14:
Thanks. The errors are expected because the Cisco camera doesn't implement the inquiry commands available on Sony cameras. There needs to be a way to detect or select the camera model and choose the appropriate inquiry command set

MarkoS65, 2021-07-14:
Hi.
Tested a couple minutes ago. Same respond as yesterday. 
[2021-07-14 19-28-08.txt](https://github.com/glikely/obs-ptz/files/6817991/2021-07-14.19-28-08.txt)
Log file is attached but it's not cleaned upp. I hope you can use it anyway otherwise, let me know and I'll clean it from all errors regarding JPEG.
I looked at Cisco User guide for VISCA codes and for "right" "left" "up" "down" comands there is note "increment pan/tilt" so maybe diagonal is not working on all cameras. I don't must have it.
For me, I need presets to work. And maybe home button to do reset of the camera (sometimes if I'm to quick with buttons it loose it's position). Speed on motions are good to have but... 
I use PrecisionHD 1080p4x S2 or new name: P40. 
When I use Tandyberg app, the pressets work, maybe a look at it (its in GitHub) can give you some ideas.

Thank you very much for your effort! :)
BRG
Marko.S

glikely, 2021-07-14:
Another test build, should reduce the error messages. Please test

https://github.com/glikely/obs-ptz/actions/runs/1032035190

wasn-eu, 2021-07-15:
yes the errors are gone.
Speed for Pan/Tilt is working (had to disable IR Receiver on the cam, with enabled IR the speed commands are without function)

But when i tried to save a preset OBS crashed.
[Crash 2021-07-15 08-46-42.txt](https://github.com/glikely/obs-ptz/files/6820936/Crash.2021-07-15.08-46-42.txt)
[2021-07-15 08-46-06.txt](https://github.com/glikely/obs-ptz/files/6820938/2021-07-15.08-46-06.txt)

glikely, 2021-07-15:
Oops, that was a dumb bug. I forgot to check if the list was empty before trying to drop an item. Try this one:

https://github.com/glikely/obs-ptz/actions/runs/1033124344

glikely, 2021-07-15:
Here's one more to try. This one dumps the cisco HWID to the log file, or at least I hope it does.

https://github.com/glikely/obs-ptz/actions/runs/1033182212

glikely, 2021-07-15:
@MarkoS65 : Regarding presets, the Cisco manual you linked does not specify the commands for setting and recalling presets, and the Sony commands return errors. Looking at the Tandyberg app, it saves presets internally to the program, and does not seem to use a preset function on the camera.

obs-ptz relies on the camera to save and restore presets. I don't have any functionality for storing presets in the controller, and it is unlikely that I'll be able to do so in the near future.

glikely, 2021-07-15:
> Speed for Pan/Tilt is working (had to disable IR Receiver on the cam, with enabled IR the speed commands are without function)

@MarkoS65 , can you check if disabling IR control fixes the pan/tilt speed control?

MarkoS65, 2021-07-15:
Hi.
Tried with both - IR_Output_Off 8x 01 06 08 03 ff and - IR_CameraControl_Off 8x 01 06 09 03 ff, but speed for Pan /tilt don't working.
Don't bother you to try more regarding my P40. I'll try to fix Tandyberg to control 2 cameras.

Very appreciated for the atempt.

BRG
Marko.S

wasn-eu, 2021-07-15:
IR_CameraControl_Off 8x 01 06 09 03 ff  worked for my PHD1080p12x for turning IR control off and pan/tilt speed working

glikely, 2021-07-15:
@MarkoS65 is speed control working with the Tandyberg app? If so, can you get a serial IO log from the app and post it here? Then I can compare with the commands I'm generating.

@wasn-eu , could you please attach the output log from the latest test build? I want to check the camera ID output

wasn-eu, 2021-07-15:
thats the log output with the hwid

[2021-07-16 00-53-08.txt](https://github.com/glikely/obs-ptz/files/6826632/2021-07-16.00-53-08.txt)

glikely, 2021-09-02:
I haven't heard anything further on this camera. If you're still having problems with the latest release then please reopen the issue

closed 2021-09-02 by glikely

KickASS0r, 2021-11-10:
> IR_CameraControl_Off 8x 01 06 09 03 ff worked for my PHD1080p12x for turning IR control off and pan/tilt speed working

How du you send commands? If i open a serial connection i can not type anything in putty.

glikely, 2021-11-10:
The commands are in binary instead of plain text. You aren't able to use putty to send the commands.

I've heard good things about 232analyser for Windows, but I haven't used it. You might be able to try serial commands using that tool: https://www.commfront.com/collections/advanced-serial-protocol-analyzer