glikely/obs-ptz #23: Adding Pelco-D

LuukVerhagen, 2021-07-13:
Now we have added Pelco-P, we should also add Pelco-D. If you haven't implemented it when I'm back from my holidays, I will pick up this Issue.

glikely, 2021-07-13:
That would be awesome, thanks. I've been looking at the code, and there is more similarities than differences between the different serial protocols in terms of code structure. The major differences that need to be accounted for are:

- datagram framing; each protocol uses a different framing delimination character
- slots: some VISCA devices have command slots that allow multiple commands in flight at once. PELCO does not
- the actual commands (obviously)

Before adding more protocols I'd like to have the refactoring work you suggested done to have a common UART handler base class.

glikely, 2021-08-02:
I've done some refactoring to share the UART code now which you can use when implementing Pelco-D

LuukVerhagen, 2021-09-01:
For your knowledge, I've already written the Pelco-D code. I only haven't able to test it. This evening I have a meeting in our church, so hopefully I am able to test the code this evening.

glikely, 2021-09-01:
Go ahead and post it for review anyway.

LuukVerhagen, 2021-09-25:
Still not able to test it. I will just open en pull-request, so you can use it if you want

closed 2021-09-25 by glikely