glikely/obs-ptz #30: VISCA over IP - UDP vs TCP/IP

Mazecreator, 2021-07-30:
Hi @glikely,

I have been talking to one of the engineers for a camera that had problems with VISCA over IP to see if there is a problem with the firmware.  He ran a few tests and showed he got the same responses via VISCA over IP as he did via Serial.  The thing was, he said you need to use TCP/IP sockets and not UDP.  I checked the commit to add VISCA over IP to this project, it looks like the project is using UDP sockets.  More reliable handling of queries might be achieve by using TCP/IP (or make this selectable as some MFG might need UDP).

Sreial:
[image]
[image]

TCP/IP:
[image]
[image]

So maybe we can add the port type to the plugin?

Great work on this needed OBS tool though as well as the integration into Face-Tracker is great for the PTZ cameras.

-Greg

Mazecreator, 2021-09-02:
I did manually confirm that the Jimcom Brand cameras only send a response back to the TCP/IP (over the bound connection) and can't reply back to the UDP packet.  I found a great development tool which can be used called Packet Sender:
https://packetsender.com/download#show

Here is the screen shot of the HOME command with TCP vs UDP transports:
[image]

Antonio0307, 2022-05-11:
我想问下，是不是所有支持VISCA协议的摄像头，都有支持网络visca协议呢。

commit a3ee68ab (2021-10-21): Add VISCA over TCP support

Initial support for cameras using VISCA-over-TCP (PTZOptics, Jimcom, and
others). This implementation works, but it is most likely buggy. Please
test and report.

Fixes: #30