glikely/obs-ptz #32: Control pan and tilt speed

eliasful, 2021-08-12:
Looking at the code, I noticed that the zoom method takes the zoom parameter and the zoom speed parameter, and it works perfectly. However, it seems that pan and tilt do not have this parameter. Am I mistaken or not? If not, is there any camera limitation for implementing this? If not again, is there a possibility that implementation will happen in the future?

glikely, 2021-08-22:
Pan and tilt do use a speed parameter. What functions are you looking at? The vertical slider between the control buttons and the preset list controls the speed, and that value is passed into `PTZDevice::pantilt()`

Here's the VISCA implementation of pantilt():
```
void PTZVisca::pantilt(double pan_, double tilt_)
{
        int pan = (pan_ > 1 ? 1 : (pan_ < -1 ? -1 : pan_)) * 0x18;
        int tilt = (tilt_ > 1 ? 1 : (tilt_ < -1 ? -1 : tilt_)) * 0x14;
        send(VISCA_PanTilt_drive, {pan, -tilt});
}
```

pantilt() is called by `PTZControls::setPanTilt()`, which uses the value of `ui->speedSlider->value()` to control the speed:

```
void PTZControls::setPanTilt(double pan, double tilt)
{
        double speed = ui->speedSlider->value();
        PTZDevice *ptz = currCamera();
        if (!ptz)
                return;

        if (QGuiApplication::keyboardModifiers().testFlag(Qt::ControlModifier)) {
                ptz->pantilt(pan, tilt);
        } else if (QGuiApplication::keyboardModifiers().testFlag(Qt::ShiftModifier)) {
                ptz->pantilt_rel(pan, - tilt);
        } else {
                ptz->pantilt(pan * speed / 100, tilt * speed / 100);
        }
}
```

What kind of camera are you working with, and can you give more detail on the behaviour your seeing with different settings of the speed slider?

eliasful, 2021-08-25:
I understood! Sorry for the inconvenience.

My camera is a 10x generic USB PTZ. I can communicate with her via VISCA Serial.

I can also control the zoom speed, slower or faster, but I can't control the pan and tilt speed.

In camera settings, I can change the pan and tilt speed with values ​​from 1 to 8, with 1 being the slowest and 8 being the fastest.

This is my camera, unfortunately they don't provide a very elaborate manual. https://www.amazon.com.br/C%C3%A2mera-Rob%C3%B3tica-PTZ-NEOiD-Zoom/dp/B098LPKVPY

glikely, 2021-08-31:
Please do a test run of OBS where you try to pan/tilt at both fast and slow speeds, and then attach the OBS log file.

eliasful, 2021-09-05:
[2021-09-05 19-41-44.txt](https://github.com/glikely/obs-ptz/files/7112466/2021-09-05.19-41-44.txt)

here I tested both slow and fast zoom. It worked correctly. I also tested the pan and tilt, but it doesn't change the speed, regardless of the speed set.

eliasful, 2021-09-05:
here are the records of the new version

[2021-09-05 19-52-17.txt](https://github.com/glikely/obs-ptz/files/7112470/2021-09-05.19-52-17.txt)

glikely, 2021-09-16:
I suspect this is a limitation of your camera. Have you found any other VISCA controller that is able to adjust the pan/tilt speed?

closed 2021-09-25 by glikely