glikely/obs-ptz #34: Enhancement: Camera IP Address Discovery Feature

Mazecreator, 2021-08-28:
Use the ONVIF protocol (not sure if VISCA over IP has such a feature) to auto discover the IP Address of the cameras on a network. If possible, it would be good to select the network if possible as some PCs might have 2 or more networks configured or multiple LAN cards. Sometimes the Autodiscovery feature runs on the wrong network and doesn't detect the desired cameras.

glikely, 2021-08-28:
Good suggestion. Unfortunately I don't have any devices that use ONVIF, so I won't be able to work on this right away. I'm happy to accept patches from anyone who gets it working though.

Mazecreator, 2021-09-04:
Sounds good, I have looked at the code prior when looking to test TCP/IP on the VISCA over IP and it seemed like a massive task to convert the UDP protocol to TCP/IP under QT (I am not that familiar with that library yet). I think ONVIF is TCP/IP, so it might fall into the same category, but might be easier from scratch.  I will look at it as I think it would be helpful.

I have done a little looking at a separate C++ TCP/IP library, but can't find anything that is light-weight...

glikely, 2026-10-01:
Fixed in #383