glikely/obs-ptz #36: 0.7 Rc1 issues

iesvme, 2021-08-29:
1. Right click PTZ Controls bar does not respond, no menu.
2. "Save presets" and "Clear presets" don't work.
3. Cameras bar has no right click menu.

glikely, 2021-08-29:
What kind of camera are you using? Is it connecting over PELCO-P or VISCA?

iesvme, 2021-08-29:
sorry, I mean just here:
[image]

> What kind of camera are you using? Is it connecting over PELCO-P or VISCA?

glikely, 2021-08-31:
Thanks for the bug report. A right-click context menus has not been implemented in the camera list, which is why you're not seeing one. 😄 

What are you expecting from "save preset" and "clear preset"? The way the plugin currently works there are a fixed number of presets named "Preset 1" to "Preset 16". Save preset and clear preset doesn't change the number of presets in the list. Instead, save preset updates the selected preset with the current camera position, and clear preset resets the name and tells the camera to forget the preset position.

iesvme, 2021-08-31:
> Thanks for the bug report. A right-click context menus has not been implemented in the camera list, which is why you're not seeing one. 😄
>
> What are you expecting from "save preset" and "clear [...]

Oh, now I see. That's a good way of it. I thought it would pop up a dialog to save the preset file on the computer or clear the preset file. Bye the way, if I needn't use 16 presets, how can I delete them?

glikely, 2021-08-31:
> Oh, now I see. That's a good way of it. I thought it would pop up a dialog to save the preset file on the computer or clear the preset file. Bye the way, if I needn't use 16 presets, how can I delete them?

Right now you can't. The list is fixed because that was the easiest way to implement it while managing the limited number of save slots on the camera. However, I do think it would be better to be able to dynamically add/remove presets and let the plugin manage the preset number, or even store the position in the configuration file for cameras that don't have preset save slots. There is already an issue tracking this.

#24

closed 2021-08-31 by glikely

iesvme, 2021-08-31:
Thank you very much! I am expecting the next version to be released.