glikely/obs-ptz #37: Focus and VISCA over IP

Mazecreator, 2021-08-30:
Love the addition of the Focus! With this version I am starting to see "Stuck" controls again while using the PTZ and Focus controls. I think I saw this back at 0.4.0 or earlier I think when you first introduced the VISCA over IP.

I experienced a little unexpected behavior with the Focus controls. I have been trying to figure out the best way to reproduce it but it is elusive in nature. It seems like at first setup, the initial state of the "Focus" controls are not known. Due to this, altering the focus manually doesn't work until you first select AF. I don't know the best way to set the One-Touch Focus, but currently that doesn't do anything I can see yet when I hit that button in Manual focus mode.

The manual focus is a great addition to the Plugin for those with obstacle in their field of view (Poles, etc.) as well as for simple artistic blur of the screen at times. Love it!

glikely, 2021-08-30:
The stuck controls is odd. I cannot think of anything that would cause that bug to reappear. The pan/tilt/zoom commands should always go out immediately without waiting for a response from the camera. Open the log window and see it the commands are going out immediately when you click and release the buttons. Are you getting any timeout messages?

glikely, 2021-08-30:
Please attach your log file. It could be that too many inquiry messages are getting sent.

Mazecreator, 2021-08-30:
I setup a clean run, towards the end of this it panned and didn't stop:
https://obsproject.com/logs/PAz25I9yMP3kORcG

glikely, 2021-08-31:
One big thing I see in the logs is there are no responses coming back from the cameras. The UDP messages are going out, but nothing comes back. I've seen this before when a camera is in a different subnet from OBS. Or, they could be being blocked by Windows firewall.

This would also be why you are seen odd behaviour on autofocus mode.

Do you know how to use wireshark to capture the network packets? If so, can you capture all of the VISCA UDP packets and see if your PC us receiving any packets from the camera?

glikely, 2021-08-31:
Here's what I see at the end of the log:

```
19:41:49.629: ViscaUDPSocket::send():858: VISCA UDP --> 01:00:00:09:00:00:00:23:81:01:06:01:00:00:03:03:ff
19:41:50.205: ViscaUDPSocket::send():858: VISCA UDP --> 01:00:00:09:00:00:00:24:81:01:06:01:0c:00:01:03:ff
19:41:51.107: ViscaUDPSocket::send():858: VISCA UDP --> 01:00:00:09:00:00:00:25:81:01:06:01:00:00:03:03:ff
19:41:52.059: ViscaUDPSocket::send():858: VISCA UDP --> 01:00:00:05:00:00:00:26:81:01:06:04:ff
19:41:53.194: ViscaUDPSocket::send():858: VISCA UDP --> 01:00:00:09:00:00:00:27:81:01:06:01:00:0a:03:01:ff
```
These commands are:
1) stop
2) pan left
3) stop
4) home
5) tilt up

I'm assuming you sent the 'home' command after a runaway pan left. The plugin did send a stop command, but the camera doesn't seem to have received it (but it is hard to tell because there are no responses received from the camera). I also notice that the stop/pan/stop sequence is within a 500ms window. Most cameras seem to have 2 command slots, which don't free up until the command completion message is sent.

What might be happening is that both slots are filled with the stop and pan commands, and then the third stop command gets dropped.

glikely, 2021-08-31:
I've held off trying to implement VISCA over TCP since I don't have any hardware that implements it. It will be very difficult to debug without a test device.

@norihiro if you've got it working then please send a pull request! 😄

glikely, 2021-09-02:
I'm closing this issue for the time being. If you're able to get responses from the camera sorted out and still see the issue then please reopen it and post the log file

closed 2021-09-02 by glikely

Mazecreator, 2021-09-02:
Sounds good, I think this will be resolved with TCP/IP from what @norihiro tells me so this feature request once filled will probably fix the problem:
https://github.com/glikely/obs-ptz/issues/30