glikely/obs-ptz #40: Only partially works with Sony BRC-300 Camera

jerburgess, 2021-09-10:
I am attempting to use this plugin with a Sony BRC-300 camera.  The plugin detects the camera, but the only four directional buttons that respond are the diagonal ones (up and left, down and left, up and right, down and right), and the home button.  When you press any of these, they don't go just a bit, they go all the way until the camera reaches it's limits, although the speed control does affect the speed at which it travels.  Any of the other buttons throw a VISCA error in the log file.

I am using the precompiled Windows download that is available through the OBS website, on Windows 10.

Here is the log file:
21:07:39.879: CPU Name: Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz
21:07:39.879: CPU Speed: 2492MHz
21:07:39.879: Physical Cores: 2, Logical Cores: 4
21:07:39.879: Physical Memory: 8075MB Total, 3007MB Free
21:07:39.879: Windows Version: 10.0 Build 19043 (release: 2009; revision: 1165; 64-bit)
21:07:39.879: Running as administrator: false
21:07:39.879: Aero is Enabled (Aero is always on for windows 8 and above)
21:07:39.880: Windows 10 Gaming Features:
21:07:39.880: 	Game DVR: On
21:07:39.883: Sec. Software Status:
21:07:39.888: 	Norton 360: enabled (AV)
21:07:39.888: 	Microsoft Defender Antivirus: disabled (AV)
21:07:39.889: 	Norton 360: enabled (FW)
21:07:39.889: 	Windows Firewall: disabled (FW)
21:07:39.889: Current Date/Time: 2021-09-09, 21:07:39
21:07:39.889: Browser Hardware Acceleration: true
21:07:39.889: Portable mode: false
21:07:40.444: OBS 27.0.1 (64-bit, windows)
21:07:40.444: ---------------------------------
21:07:40.479: ---------------------------------
21:07:40.479: audio settings reset:
21:07:40.479: 	samples per sec: 48000
21:07:40.479: 	speakers:        2
21:07:40.482: ---------------------------------
21:07:40.482: Initializing D3D11...
21:07:40.482: Available Video Adapters: 
21:07:40.485: 	Adapter 0: Intel(R) HD Graphics 3000
21:07:40.485: 	  Dedicated VRAM: 33554432
21:07:40.485: 	  Shared VRAM:    1711276032
21:07:40.485: 	  PCI ID:         8086:126
21:07:40.486: 	  Driver Version: 9.17.10.4459
21:07:40.486: 	  output 0: pos={0, 0}, size={1366, 768}, attached=true, refresh=60, name=
21:07:40.493: Loading up D3D11 on adapter Intel(R) HD Graphics 3000 (0)
21:07:40.499: D3D11 loaded successfully, feature level used: a100
21:07:40.499: DXGI increase maximum frame latency success
21:07:41.539: ---------------------------------
21:07:41.539: video settings reset:
21:07:41.539: 	base resolution:   720x480
21:07:41.539: 	output resolution: 720x480
21:07:41.539: 	downscale filter:  Bicubic
21:07:41.539: 	fps:               30/1
21:07:41.539: 	format:            NV12
21:07:41.539: 	YUV mode:          709/Partial
21:07:41.540: NV12 texture support enabled
21:07:41.541: Audio monitoring device:
21:07:41.541: 	name: Default
21:07:41.541: 	id: default
21:07:41.546: ---------------------------------
21:07:41.547: Skipping module '../../obs-plugins/64bit/chrome_elf.dll', not an OBS plugin
21:07:41.557: [CoreAudio encoder]: CoreAudio AAC encoder not installed on the system or couldn't be loaded
21:07:41.558: Failed to load 'en-US' text for module: 'decklink-captions.dll'
21:07:41.560: Failed to load 'en-US' text for module: 'decklink-ouput-ui.dll'
21:07:41.638: [AMF] Unable to load 'amfrt64.dll', error code 126.
21:07:41.643: [AMF] AMF Test failed due to one or more errors.
21:07:41.643: Failed to initialize module 'enc-amf.dll'
21:07:41.659: Skipping module '../../obs-plugins/64bit/libcef.dll', not an OBS plugin
21:07:41.674: Skipping module '../../obs-plugins/64bit/libEGL.dll', not an OBS plugin
21:07:41.676: Skipping module '../../obs-plugins/64bit/libGLESv2.dll', not an OBS plugin
21:07:41.688: [obs-browser]: Version 2.14.3
21:07:41.688: [obs-browser]: CEF Version 75.1.16+g16a67c4+chromium-75.0.3770.100
21:07:41.697: [noise suppress: Nvidia RTX denoiser disabled, redistributable not found]
21:07:42.477: Failed to load 'en-US' text for module: 'ptz-controls.dll'
21:07:42.477: PTZ Controls plugin v0.7.0
21:07:42.640: gamepads found 0
21:07:42.640: ViscaUART::get_interface():735: Looking for UART object COM4
[... 372 lines cut ...]
21:12:40.022: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:40.022: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:40.075: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:13:00:02:03:ff
21:12:40.104: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:40.104: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:40.135: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:00:00:03:03:ff
21:12:40.168: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:40.168: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:40.220: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:13:00:02:03:ff
21:12:40.253: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:40.254: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:40.291: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:00:00:03:03:ff
21:12:40.320: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:40.321: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:46.119: gamepads found 0
21:12:48.907: ViscaUART::get_interface():735: Looking for UART object COM5
21:12:51.294: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:13:10:02:01:ff
21:12:51.326: ViscaUART::receive_datagram():691: VISCA <-- 90:41:ff
21:12:51.342: ViscaUART::receive_datagram():691: VISCA <-- 90:51:ff
21:12:51.443: PTZUARTWrapper::send():91: COM5 --> 81:01:06:01:00:00:03:03:ff
21:12:51.477: ViscaUART::receive_datagram():691: VISCA <-- 90:60:02:ff
21:12:51.477: PTZVisca::receive():547: VISCA PTZ received error: 90:60:02:ff
21:12:52.399: PTZUARTWrapper::send():91: COM5 --> 81:01:06:04:ff
21:12:52.506: ViscaUART::receive_datagram():691: VISCA <-- 90:41:ff
21:12:53.482: ViscaUART::receive_datagram():691: VISCA <-- 90:51:ff

glikely, 2021-09-16:
I've got a theory. It seems to be throwing an error whenever speed is set to '0' in the pan/tilt commands. This would happen in any non-diagonal move, and also on the stop command. As an experiment I've made a change so that the speed field can never be zero. Please try it out and let me know if it fixes the problem. You can find the test build here:

https://github.com/glikely/obs-ptz/actions/runs/1241193296

jerburgess, 2021-09-19:
Yes!  That was it!  It works perfectly now.

commit 557ee826 (2021-09-16): VISCA: Ensure speed argument is never 0

Some cameras (i.e., BRC-300) don't like the speed field set to '0', even
if the command is to stop motion. Ensure that the speed field is set to
a minimum of 1 so that the camera doesn't throw an error.

Fixes: #40
commit 951fb780 (2021-09-19): VISCA: Ensure speed argument is never 0

Some cameras (i.e., BRC-300) don't like the speed field set to '0', even
if the command is to stop motion. Ensure that the speed field is set to
a minimum of 1 so that the camera doesn't throw an error.

Reported-by: https://github.com/jerburgess
Fixes: #40
closed 2021-09-19 by glikely