glikely/obs-ptz #41: Make size of gamepad adjustable

Bennoholunder, 2021-09-10:
The gamepad of the ptz controls dock is - too small - on a 4K Monitor to be operated easily.
It would be very much appreciated if the ratio between the different sections of the gamepad could be adjusted according to the individual needs. Either by dragging the border between the sections or by entering a ratio factor in the settings dialogue.
Thanks for consideration!

glikely, 2021-09-12:
Please give this test build a try. This is just a test. There are still layout problems, but I want to get your feedback on how it works for you.

https://github.com/glikely/obs-ptz/actions/runs/1225424572

Bennoholunder, 2021-09-12:
The test version with the adjustable size of the control buttons works great on my side:
( I compiled the ptz-controls on openSUSE 15.3 and I am using VISCA over IP ).
I fiddled around a bit with the ptz-controls.ui with QtDesigner and found the attached layout even more handy.
I placed the zoom-in and -out buttons right of the arrow-pad and changed the arrow icons to a hollow shape so they don't stick out too bold and bright. See my screenshot.
[image]
I would also recommend to change the icon size calculation from ( - 2*10 px ) to ( * 3 / 4 ) of qMin (height, width) to scales proportional to the button size.

glikely, 2021-09-16:
I've reworked the layout as suggested and made sure all the pan/tilt buttons remain square.

closed 2021-09-16 by glikely