glikely/obs-ptz #44: Ability to set IP camera address/ID (not IP)?

normen, 2021-09-20:
Hi,

first of all thanks for this software. I was wondering if it was possible to set the VISCA ID/address for the IP option. I am using a UDP to RS485 converter and have multiple cameras daisy chained to it, only the first camera seems to react.

Thanks again,
Normen

glikely, 2021-09-20:
The current IP support in the plug-in is based on Sony's VISCA over IP design which doesn't allow for multiple cameras at an address. Using a UDP to Serial adapter is a different configuration. Probably should be doable, but I'll need more information. Can you post your OBS log file? Also, what is the UDP to RS485 device you're using?

normen, 2021-09-21:
The adapter is a self-built Arduino/ESP32 based device with a RS485 converter, the code is an adapted version of this: https://github.com/utopiantools/visca-ip-proxy-arduino - I changed the code to work on an ESP32 and removed the address set commands (which don't work for my cameras anyway as they set the address/ID via DIP switches).

This seems to work for multiple cameras when I send VISCA commands directly via code, with different addresses/IDs than `0x81`, which I think would be the only thing that needs to be changed in obs-ptz to support such a configuration.

glikely, 2021-09-21:
Very cool! Have you posted your modified code to GitHub? I'm keen to take a look, and it is something I could make use of too.

normen, 2021-09-21:
> Very cool! Have you posted your modified code to GitHub? I'm keen to take a look, and it is something I could make use of too.

No I didn't as I am not sure if I'll make more changes but I basically just removed the tally/LED functionality (as its tied to the M5 platform) and otherwise used the code as is.

commit 760a0472 (2025-05-05): visca/udp: Allow camera to be specified by hostname

Change the Visca over IP driver to enable a hostname to be used to
connect to the camera in addition to IP addres.

Fixes: #44