glikely/obs-ptz #45: Program max speed for presets on Sony BRC-300

jerburgess, 2021-09-22:
This is not a pressing issue, but when you tell the Sony BRC-300 to recall a preset, it goes rather quickly to the preset.  Apparently, there is a way to set the speed at which the camera moves to a preset if you're using the Sony RM-B300 controller.  I'm guessing there is a way to send it through the VISCA interface as well.  Here's the appropriate section in the controller manual.  Is it possible to implement this in your software as well?

[image]

glikely, 2021-09-25:
I cannot find documentation on how to control preset recall speed. I’ll need a serial trace between a controller and camera that includes the commands if I’m to add support for this. Unfortunately the BRC-300 doesn’t have protocol details published.

I’m closing this issue for now due to insufficient data, but please reopen if you can find the technical details.

glikely, 2021-09-25:
Hold on, I just found this link which includes that camera’s commands:

https://www.snapav.com/wcsstore/ExtendedSitesCatalogAssetStore/attachments/documents/USBProducts/ManualsAndGuides/Visca%20Protocol2.pdf