glikely/obs-ptz #48: Cannot control serial camera with Linux plugin

gillyboatbruff, 2021-09-27:
I have a laptop where I dual boot Windows and Linux.  Using Windows, I am able to use the plugin to control the camera with no issues.  I complied the plugin for Fedora 34, and am not able to control the camera at all.  I am attaching logs for both Windows and Linux
[2021-09-26 11-11-04_windows.txt](https://github.com/glikely/obs-ptz/files/7232498/2021-09-26.11-11-04_windows.txt)

[2021-09-26 12-31-41_linux.txt](https://github.com/glikely/obs-ptz/files/7232499/2021-09-26.12-31-41_linux.txt)
.

glikely, 2021-10-17:
Double check the address setting in the config. From the log files I can see that on Linux the device address being sent out is '0' (First byte of datagram == 0x80) but on Windows it is correctly set to '1' (First datagram byte == 0x81)

gillyboatbruff, 2021-10-19:
That was enough to help me pin down the problem.  Inside OBS, the plugin showed that it was set to VISCA ID 1, but it still wasn't working.  I checked the ptz-controls/config.json file directly, and there the ID was set to 0.  I changed it to 1, restarted OBS, and I can now control the camera in Linux as well.  Thanks for the help.