glikely/obs-ptz #65: Speed Ramping

EaB11, 2021-11-07:
it would be good to have speed ramping controls to avoid the jerky movement of many PTZ cameras...  so movement could begin slow and then within a second or two increases to the maximum speed..  this would help PTZ movement to be used within the production more, rather than just a way of changing between static shots... 

ideally it would be good to have independent ramp controls for each axis (pan/tilt and zoom).. for example you might want slower speeds for pan/tilt controls but prefer zoom speed a nudge faster.. 

also would be good to have the ramp settings apply when the camera is the active output but then have separate speed controls for repositioning when not the active camera..  you generally want preview reposition to happen quickly and the smoother ramping for when camera is active... 

finally it would be good if ramp settings were able to be set individually for each camera.. that way if you had different PTZ models that behaved slightly different they could be calibrated to match..

arteVact, 2022-07-13:
That would be really fantastic. A real reduction in workload and productions would become much more professional!

glikely, 2022-11-25:
Please try out v0.13.0-test2. It implements speed ramping

closed 2022-11-25 by glikely

arteVact, 2022-11-28:
Awesome, thanks a lot!!!

commit afcf7cc8 (2022-11-25): Implement speed ramping

Replace the UI speed slider with a time-based speed ramp. Pan/tilt/zoom
movements will start slow, and ramp up to faster speeds the longer the
button is held down.

The slider is entirely removed from the UI now as unnecessary. Users can
use the <ctrl> or <shift> modifies to get fast and slow movement
respectively.

Fixes #65