glikely/obs-ptz #67: obs_ptz lib on MacOS failed to load

AllenVM, 2021-11-09:
Looks like I'm missing a required library.  Assume this should be part of the binary distribution?

- Allen

Relevant OBS Log entries:

08:53:53.897: os_dlopen(/Library/Application Support/obs-studio/plugins/ptz-controls/bin/ptz-controls.so->/Library/Application Support/obs-studio/plugins/ptz-controls/bin/ptz-controls.so): dlopen(/Library/Application Support/obs-studio/plugins/ptz-controls/bin/ptz-controls.so, 0x0101): Library not loaded: /tmp/obsdeps/lib/QtCore.framework/Versions/5/QtCore
08:53:53.897:   Referenced from: /Library/Application Support/obs-studio/plugins/ptz-controls/lib/QtSerialPort
08:53:53.897:   Reason: tried: '/tmp/obsdeps/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/System/Library/Frameworks/QtCore.framework/Versions/5/QtCore' (no such file)
08:53:53.897: 
08:53:53.897: Module '/Library/Application Support/obs-studio/plugins/ptz-controls/bin/ptz-controls.so' not loaded

glikely, 2021-11-09:
Looks like I need to do more work on packaging so that it searches for the Qt libraries in the OBS studio path.

AllenVM, 2021-11-09:
Thanks.  So, are the QT Libraries are a prerequisite?  I'm not sure which distribution on GitHub to use.  Does licensing allow them to be included in your distribution?

As an aside, we're using your library already for controlling a Minrray PTZ Camera connected to OBS via NDI over Ethernet.  Your OBS_ptz lib worked great on a Windows machine, but our main machine for streaming broadcasts is a Mac, hence our interest in your port to MacOS - it is a huge benefit!  Thanks.

glikely, 2021-11-09:
Yes Qt is required. OBS studio uses the Qt widget library. For this plugin I added in the Qt serial port library, but apparently the way I packaged it means it cannot find the core libraries that are shipped with OBS. This is a configuration issue that I need to sort out.

Glad to hear the plug-in is working well for you.

glikely, 2021-11-10:
Should be fixed in v0.10.2. Please test and report back

glikely, 2021-11-10:
https://github.com/glikely/obs-ptz/releases/tag/v0.10.2

closed 2021-11-10 by glikely

AllenVM, 2021-11-10:
Loaded with no errors, was able to make it visible via "Dock" menu.  Will test it out fully when I am onsite and have access to our PTZ Camera.

AllenVM, 2021-11-11:
Grant:

I just fully exercised the 0.10.2 release of obs_PTZ lib on my Mac.   Works great, no problems!

-Allen Vander Meulen

glikely, 2021-11-11:
Awesome! Glad it is working for you