glikely/obs-ptz #71: UVC & onvif PTZ support

tconley, 2021-11-21:
Can there be support for UVC and onvif cameras? I've got a budget to support author or anyone else bounty-style for UVC

glikely, 2021-11-21:
I've already got a feature request for ONVIF support in issue #16, but nobody has asked for UVC support yet. Thanks for filling this. I'll rename this issue to specifically request UVC support.

https://github.com/glikely/obs-ptz/issues/16

I'm happy to add support if someone with the hardware gets it implemented. Someone will need to send me equipment if I'm the one to do it.

iam-TJ, 2022-08-08:
As a newcomer to OBS - installed just now on Debian Bullseye, if given some guidance I could probably contribute here. I installed this plug-in thinking it already supported UVC because I was searching for a UVC PTZ controller option.

Currently those controls are available as part of the video camera controls under Source > Device > Properties. The problem with using that is two-fold:

1.  As soon as the Properties dialog opens it resets the PTZ values causing the camera to change its aim and zoom to the 'zero' point - in my case turns away through -150 degrees pan, and +30 degrees tilt!
2. The controls are exposed as sliders that are hard to adjust smoothly.

In the case of the camera I'm testing there are several UVC controls exposed that affect PTZ so I suspect we'd need some way to map those to the plug-in controls as a minimum, and probably allow some kind of user scaling of the control sensitivity.

In the case of the camera I have here the PTZ-related UVC controls exposed are:
```
Pan, Absolute (range -612000 - 612000)
Tilt, Absolute (range -108000 - 324000)
Focus, Automatic Continuous (Boolean)
Zoom, Absolute (range 0 - 16384)
Zoom, Continuous (range 0 - 7)
Pan, Speed (range 0 - 24)
Tilt, Speed (range 0 - 24)
```

glikely, 2022-09-15:
I experimented with libuvc, and was able to create a simple command line utility for sending PTZ commands to a UVC device. It would require getting libuvc linked into obs-ptz, and making sure it works on all three OSes.

I would definitely appreciate the help if you or someone else can implement it.

iam-TJ, 2022-09-15:
Like everyone else I have a stack of things in my ToDo list but I'll add this to the pile. If anyone else wants to dive into it in the meantime let us know so we don't duplicate work!

EtienneNoir99, 2023-01-19:
I would be very grateful if UVC cameras are also supported. We use a Logitech PTZ Pro 2 and control it with this little tool written in C++: https://github.com/xMRi/PTZControl

However, I would like to switch to OBS and create scenes that use specific presets.

gr3m1in, 2024-01-18:
It would be really great to have UVC control for my [TELESIN ACM-SC1-C2K](https://www.bhphotovideo.com/c/product/1714912-REG/telesin_acm_sc1_c2k_aicoco_smart_live_stream.html/overview)

glikely, 2025-04-30:
USB UVC support is now in mainline

closed 2025-04-30 by glikely