glikely/obs-ptz #73: Debian 11 building fail

ogmkp, 2021-12-14:
Hi.
I wanted to update obs-ptz 0.5.0 to 0.14.4 on Debian 11, but impossible to compile from v0.9.0. There are gcc compatibility errors.
Can you tell which gcc version or distribution is compatible from which versions?
If possible, how to build for Debian 11 stable ?

Thank you.

glikely, 2021-12-14:
Please attach the failure you are seeing from gcc

ogmkp, 2021-12-15:
In build/UI/frontend-plugins/obs-ptz/ I have the following files:
```
.
..
CMakeFiles
cmake_install.cmake
config.generated.h
Makefile
obs-studio_autogen
obs-studio.so
ptz-controls.qrc.depends
qrc_ptz-controls.cpp
ui_ptz-controls.h
ui_settings.h
```

I clone the repo in UI/frontend-plugins/obs-ptz and I add "add_subdirectory(obs-ptz)" in UI/frontend-plugins/CMakeLists.txt.
Do you think obs-studio.so might be the binary?
Am I making a mistake somewhere?

ogmkp, 2021-12-30:
Hi @glikely, do you have idea ?
The issue occurs from v0.9.0.

glikely, 2022-01-04:
I don't recommend building the plugin in the obs-studio build tree anymore. I recommend building obs-studio first, and then building obs-ptz from its own directory. Instructions are in the README file.

I've also checked if obs-ptz will build against the Debian 11 packaged version for libobs-dev. There were a few issues that I've fixed, and now you should be able to build obs-ptz without rebuilding obs-studio when using the Debian package. Instructions for this are also in the README.

I hope that helps. I'm closing this issue, but you can reopen it if you still have problems

closed 2022-01-04 by glikely

ogmkp, 2022-01-04:
Thanks @glikely 
Now, I build obs-ptz out of the box without using libobs-dev from repos because I prefer to get the last official release builded by me.
Is it not recommended to build inside obs-studio build tree for sure ?
It's really more practical for maintaining consistency.

glikely, 2022-01-04:
It is not recommended as it isn't how the plugin gets built on a regular basis, and I don't think my recent CMake changes are compatible with building it in-tree. If you can get it to work again then I'm happy to take patches to support in-tree builds.

ogmkp, 2022-01-04:
Unfortunatly, I'm not a developer but this will be great.
Thank you for this projet that is very greatfull !

ogmkp, 2022-01-07:
Last thing, is it possible for you to give already built deb amd64 for debian 11 in your release section ?