glikely/obs-ptz #77: Feature Request: OSC integration

ygreq, 2022-01-02:
Hey there! 

I would really like to use a gamepad or joystick with this plugin. I read that there are some issues with using a gamepad. Actually I did not quite understood how you can use a gamepad with this plugin in the first place. I did activate the function, but I don't know what else to do. I don't know if I am affected by the bug or I did not set something right or maybe I just don't have the right gamepad in the first place. Maybe you can guide me to a tutorial of some sorts..

But I created a gamepad to OSC patch inside of Max (Cycling'74 app). And it would be great to just send OSC data from Max to OBS PTZ plugin. Is it that hard to implement OSC? :)

Thank you so much!!

glikely, 2022-01-04:
What is needed to connect OSC data to OBS? Is it just a socket connection or something else?

ygreq, 2022-01-07:
Unfortunately I don't have an answer for this as I am not a coder. I would have forked your project if I knew how. :) I only program in Cycling 74 Max which is a visual programming language. 

From my understanding it is easy to integrate. I found a recommendation to use this library with C++: [https://code.google.com/archive/p/oscpack/](url). The recommendation is from here [https://stackoverflow.com/questions/5785341/basic-osc-tutorial-in-c-what-library-to-use](url)

And I found an integration for OSC in OBS: [https://github.com/jshea2/OSC-for-OBS](url) Maybe it can give you some pointers considering it implies the same main app. 

PS: In case you haven't had a chance to look into OSC, it is a buffed up version of MIDI. Or how the devs put it, <<Open Sound Control (OSC) is a protocol for communication among computers, sound synthesizers, and other multimedia devices that is optimized for modern networking technology. >>

ygreq, 2022-01-07:
Regarding OSCpack, here is what they say: 

Oscpack is simply a set of C++ classes for packing and unpacking OSC packets. Oscpack includes a minimal set of UDP networking classes for Windows and POSIX. The networking classes are sufficient for writing many OSC applications and servers, but you are encouraged to use another networking framework if it better suits your needs. Oscpack is not an OSC application framework. It doesn't include infrastructure for constructing or routing OSC namespaces, just classes for easily constructing, sending, receiving and parsing OSC packets. The library should also be easy to use for other transport methods (e.g. serial).

The key goals of the oscpack library are:

-Be a simple and complete implementation of OSC
-Be portable to a wide variety of platforms
-Allow easy development of robust OSC applications (for example it should be impossible to crash a server by sending it malformed packets, and difficult to create malformed packets.)>>

ygreq, 2022-01-11:
Hi, Grant! Have you managed to look into it? Is it doable?

glikely, 2022-11-10:
It is doable, but I don't have the time to do it. If you or someone else is able to write the code and test then I'm happy to merge it in.

closed 2025-04-30 by glikely