glikely/obs-ptz #78: Ability to import and export presets

tuxtlequino, 2022-01-03:
The ability to export presets to create backups or set up a new computer is key to a lot of people. I think this will be a good feature to have with this software.

glikely, 2022-01-04:
Good suggestion, thanks!

purplecho, 2026-03-05:
Is there any news on this feature request? An unrelated hardware issue means I'm trying to rapidly set up a replacement computer, and I have managed to bring across the PTZ preset *names* without any of the associated data (by migrating the entire `plugin-config` folder to the new computer) and there's no immediately apparent way to rescue the preset data. In the short term, is the data stored anywhere that it could be manually rescued?

glikely, 2026-09-20:
This feature will be part of the next release

closed 2026-09-20 by glikely