glikely/obs-ptz #79: Camera selection is lost when selecting a preset

mudnug, 2022-01-04:
Steps:
Select a camera
Double-click a preset to apply it.

Actual: ListBox items lose selected effect after preset is double clicked (on Windows with latest OBS and dark theme)

Expected: The readme shows "PTZ" with a different color so that the user is aware of which camera he is operating
[image]

glikely, 2022-11-10:
What platform are you running on, and can you attach screenshots of dock?

mudnug, 2022-11-10:
Windows.

What "dock" are you referencing?

glikely, 2022-11-10:
Sorry, I realise now that you both told me the platform was Windows, and you attached a screenshot of the controls dock. Complete failure on my side.

Technically the widgets are behaving exactly as they should. When the widget focus switches from the camera list to the preset list, or visa versa, the actively focused widget gets highlighted, and the inactive one is highlighted with a subdued colour. Unfortunately the subdued colour in the OBS dark theme is almost the same as the background colour. You can see this working by using <TAB> and <SHIFT><TAB> to move focus back and forth between the two lists.

I don't have an easy fix for this one. From my current knowledge fixing this it requires changing the theme file in OBS studio itself to give "enabled, but not focused" a different colour. I haven't figured out how to overlay my own theme on top of the OBS one, or how to find out when OBS has loaded a new theme so that the plugin can have a matching one.

I'll do some research and see if I can figure it out, but I don't know when I'll be able to fix it.

glikely, 2022-11-10:
There is a hack that you can use to fix it. If you open C:\Program Files\obs-studio\data\obs-studio\themes\Yami.qss and replace every instance of QListWidget with QListView, then you'll get the colours that you want.

mudnug, 2022-11-10:
Would you feel this is something that would need to be fixed (without a workaround) by using a different UI control for this extension, a different theme or something else?

glikely, 2022-11-10:
Best solution is to get the theme file changed in the main OBS Studio project.

glikely, 2025-04-30:
Should be fixed now in current mainline. Reopen if you still have the problem.

closed 2025-04-30 by glikely