glikely/obs-ptz #81: Feature Request: Camera Slider Controls

Nafru, 2022-01-07:
I am in the process of building a two axis camera slider for streaming aquariums, using a Raspberry Pi, Arduino, GRBLSheild and Stepper motors. 

Is it possible to build in simple G-code moves and set locations in the OBS PTZ software?

glikely, 2022-11-10:
A G-code backend would not be difficult to add. It is just another protocol. I'm happy to add it if someone writes the code and can test.

playaspec, 2023-01-27:
> Is it possible to build in simple G-code moves and set locations in the OBS PTZ software?

Why not give the slider a VISCA interface? Then it would be compatible with a _**WIDE**_ variety of professional and prosumer gear. There's a number of exiting VISCA libraries for both Arduino and  Linux.

glikely, 2023-02-21:
I'm closing this issue for now. I'd be happy to add slider support, but I agree wit @playaspec's suggestion. If you can make it receive VISCA commands then it will work out of the box. VISCA doesn't have a slider command, but you could reuse the pan/tilt command so that it maps to the existing buttons. The VISCA pan/command is the following bytes over the serial port:

```
Up        81010601vv ww0301FF
Down      81010601vv ww0302FF
Left      81010601vv ww0103FF
Right     81010601vv ww0203FF
UpLeft    81010601vv ww0101FF
UpRight   81010601vv ww0201FF
DownLeft  81010601vv ww0102FF
DownRight 81010601vv ww0202FF
Stop      81010601vv ww0303FF

vv := pan speed, 0x01..0x18
ww := tilt speed, 0x01..0x14
```

closed 2023-02-21 by glikely

glikely, 2026-10-28:
Hi Nafru,

Picking up on this old thread, did you every build your slider and get it integrated? The architecture of this plugin has changed quite a bit and it is now possible to add more movement axes. A G-Code backend would certainly be possible, although I would still recommend VISCA as an easier way to do it.