glikely/obs-ptz #82: Feature request: Support for Obsbot PTZ and other USB devices

Verrou, 2022-01-08:
Any chance we could get support for PTZ devices (such as the Obsbot PTZ cameras (Tiny, Tiny 4k, Tail etc.)) that are connected via USB? Unsure on how difficult this would be to implement, but can see that it is using generic Windows drivers and can be controlled using hand gestures or OBSBOT's proprietary software.
https://www.obsbot.com/obsbot-tiny-4k

AndreV84, 2022-02-25:
@norihiro What is the solution if any?

norihiro, 2022-02-25:
> @norihiro What is the solution if any?

If I understand correctly, the manufacture of the camera does not disclose API to control from third party application. It's unlikely for this plugin to be able to support the camera.

GarthDB, 2023-08-06:
@norihiro I found some documentation on OSC commands supported by OBSBOT https://www.obsbot.com/explore/obsbot-webcam/osc

shanness, 2023-08-06:
This works fairly well with my OBSBOT tiny 4k
https://github.com/soyersoyer/cameractrls
as does other PTZ control stuff on linux.  So seems pretty standard

glikely, 2025-04-19:
Closing this one as duplicate of the UVC enhancement request

closed 2025-04-19 by glikely