glikely/obs-ptz #83: Bug report: Visca over UDP (Birddog P100) looses connection

DachsbauTV, 2022-01-15:
Hi @glikely 

Just wanted to share a bug with you, which occurs almost every time we are using the plugin.

The controls usually work fine for a couple of minutes and then it seems like the connection to the camera is lost. The problem is, that you can not just reconnect to the camera through the plugin settings, you actually have to restart the cam to get the controls back.

At first we were able to find an issue, which lead to the issue: Whenever another connection to the cameras network connection is initialized, the plugin would stop working. In our case this was the NDI Studio Monitor or the Birddog Controls app.
When we stopped using the apps the connection would stay up for much longer.

Unfortunately now the issue started to occur again way quicker, after about 1 minute, even though there were no other apps connecting to the cam. 

I can gladly provide more logs or anything which could help you find the issue!

OBS 27.1.3.
PTZ 10.4.
Win10
Birddog P100 via PoE and HDMI

glikely, 2022-11-10:
Thanks for the bug report. Please attach a copy of the OBS log that shows when the issue occurs

glikely, 2022-11-26:
Closing due to lack of information

commit e215edeb (2022-12-02): visca: fix reconnect timer for TCP

The `QTimer::singleShot()` call for TCP reconnect was missing the
required `()` on the destination slot, causing the timer to be
ineffective. With this fix TCP reconnection should now work.

Fixes: #83
closed 2022-11-26 by glikely