glikely/obs-ptz #86: Configure more than 16 presets?

nicolasvivant, 2022-03-09:
Hello,

The number of presets seems to be limited to 16. I didn't find a way to increase it.
Is it possible?

Thanks for your help!

allegfede, 2022-03-15:
Got the same request.
Could be usefull to add and remove from the preset list the ones not used or add more than the original 16.

splaestro, 2022-05-03:
Amen, amen! We use PTZOptics cameras, which will allow for 256 presets... I don't think anyone needs _that_ many, but one of our cameras does have 20+ at present.

arek-pelski, 2022-10-01:
This would be very useful and convenient. I join in the request.

rasamassen, 2022-12-05:
Agreed. It should be easy to add/delete presets (when adding, it automatically picks the next open number) so that each camera only shows the number of presets we're actually using.

AnthonyG83, 2022-12-09:
I would like to see the ability to add more presets.

Thanks.

ClaudeQc, 2023-01-03:
Happy new Year,

I request more than 50 preset yoo if you can. And more Store preset in shotrt cut preference not only recall preset.

Thank's
[image]

glikely, 2023-01-29:
I've made the number of presets configurable and closed this issue. @ClaudeQc Please open another issue to request Memory Store shortcuts be added

closed 2023-01-29 by glikely

nicolasvivant, 2023-01-30:
Thanks a lot!

commit 4aa426a5 (2023-01-29): Make number of presets configurable

Many cameras have more than 16 presets. Make it possible to have more
than 16 in the preset list. For now this is just a simple tunable that
changes the number of presets that are preloaded into the list. The
preset system does need more rework to make it possible to reorder
presets and have preset data stored locally, but that is a job for a
future commit.

Fixes #86
commit f6176c7a (2023-01-30): Make number of presets configurable

Many cameras have more than 16 presets. Make it possible to have more
than 16 in the preset list. For now this is just a simple tunable that
changes the number of presets that are preloaded into the list. The
preset system does need more rework to make it possible to reorder
presets and have preset data stored locally, but that is a job for a
future commit.

Fixes #86