glikely/obs-ptz #91: Controls and zoom are not working on Windows 11

wesfc, 2022-04-07:
Hello, are there someone running this plugin on Windows 11?

I'm trying to use but I cannot control de camera by OBS, I just can see the image from camera, but the controls are not working.

I would like to know if this plugin already have support for Windows 11.

In OBS logs I can see some logs from plugin trying to send the command for camera, but the moviments and zoom in/out doesn't works.

I'm already running OBS as administrator.

I will attach my OBS logs.

Thanks in advance
[2022-04-07 20-24-41.txt](https://github.com/glikely/obs-ptz/files/8447477/2022-04-07.20-24-41.txt)

justinParton, 2022-09-15:
I am currently running this on a production system using windows 11.
Camera's are controlled through RS485 as VISCA serial protocols with no issues.

glikely, 2022-11-06:
I cannot see any problems in the log file, apart from the camera not responding to any of the commands (which might be normal. Some cameras don't send responses to VISCA commands. There were not any errors reported on opening the serial port. Please double check your UART settings, and that the camera is using the same UART speed as the controller. I'm closing this issue for now, but feel free to reopen when you have more information.

closed 2022-11-06 by glikely