glikely/obs-ptz #92: Build error - MacOS Monterey

wykananda, 2022-05-10:
I'm trying to build the plugin on the latest MacOS Monterey (Silicon/ARM64).  However, when I try to build the plugin, I get the error below.  I'm using the latest Master of OBS Studio which builds and runs fine.  Any guidance on how to resolve this would be appreciated!
  
```
me@JohnDoe obs-ptz % ci/macos/build-macos.sh
=> Building plugin for macOS.
CMake Deprecation Warning at /Users/JohnDoe/Documents/Development/obs-studio/cmake/external/ObsPluginHelpers.cmake:23 (message):
  You are using an outdated method of adding 'libobs' to your project.  Refer
  to the updated wiki on how to build and export 'libobs' and use it in your
  plugin projects.
Call Stack (most recent call first):
  external/FindLibObs.cmake:98 (include)
  CMakeLists.txt:43 (include)

CMake Error at /Users/JohnDoe/Documents/Development/obs-studio/cmake/external/ObsPluginHelpers.cmake:27 (find_package):
  By not providing "Findlibobs.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "libobs", but
  CMake did not find one.

  Could not find a package configuration file provided by "libobs" with any
  of the following names:

    libobsConfig.cmake
    libobs-config.cmake

  Add the installation prefix of "libobs" to CMAKE_PREFIX_PATH or set
  "libobs_DIR" to a directory containing one of the above files.  If "libobs"
  provides a separate development package or SDK, be sure it has been
me@JohnDoe obs-ptz % 
```

myemuk, 2022-09-14:
Is any solution? Same error

glikely, 2022-09-15:
I'm in the process of updating the build process to work with OBS v28, which will also address the MacOS build problems on both x86 and Apple silicon. I've currently got it building, but there are still some fixes needed before it is ready to use. That said, you should be able to build from the current main top of tree.

closed 2022-09-15 by glikely