glikely/obs-ptz #104: Support OBS 28.0

iesvme, 2022-09-02:


bpmacedo, 2022-09-02:
+1 here

Thx 4 the support!!!

hmsjel, 2022-09-05:
Is there comming an update to  OBS 28

pviall, 2022-09-06:
Another one who updated OBS before I should have...

palaniappan1989, 2022-09-06:
+1 
Thanks for the support.

glikely, 2022-09-14:
Hi all. I've pushed out experimental support for v28. It has only been tested on an Apple Silicon Mac. Linux and x86 Mac have been built, but not tested. Windows builds are not yet working. Also connection over serial is disabled because the QSerialPort library is missing.

Images aren't available yet. I need to do some more work on the build environment to get those working.

The changes to Qt6 required moving up to the latest version of the obs-plugintemplate, but my code had deviated enough that it broke how additional libraries are brought in. QGamePad is gone entirely, and QSerialPort needs to be re-integrated.

ThomasConnolly, 2022-09-14:
Thanks, glikely, for the work you put into this plugin. I wish I could help, but I'm not a coder.

glikely, 2022-09-24:
Fixed in v0.12.0-pre5

closed 2022-09-24 by glikely