glikely/obs-ptz #106: Hotkey: Zoom Speed and Program/Preview Tabs

justinParton, 2022-09-15:
Hi!

First: love the plugin and all that you have accomplished so far!

Is it possible to get hotkey's for the zoom speed and "Program" and "Preview tabs?  My team is constantly having to reach for the mouse and click those buttons. 

regardless -- still a fantastic plugin!
Justin

glikely, 2022-11-08:
I've added hotkeys for program/preview auto selection and for toggling the move lock, and you can expect to see those land in the next release. However, experimenting with hotkeys on the speed slider hasn't resulted in anything I'm happy with. having hotkeys to increase/decrease the speed was either too slow, or too imprecise. Having hotkeys to jump straight to slow/medium/fast might work, but feels clunky.

What behaviour do you want from speed change hotkeys? i.e., what do you want the workflow to be?

glikely, 2022-11-10:
I played around some more and found some speed increase/decrease behaviour that isn't bad, so I've gone ahead and added them too.

closed 2022-11-09 by glikely

justinParton, 2022-11-13:
Checked with my team and they said percentage changes would better than none. One of them said that a hotkey slider increase of 5% would be effective. Might not be smooth, but better than a manual change using a mouse.

commit 4551ddb2 (2022-11-09): Add hotkeys for autoselect, speed, and unlock

Register hotkey options for choosing follow program, follow preview,
changing the pan/tilt/zoom speed, and unlock controls actions.

Fixes #106