glikely/obs-ptz #120: scale the camera control

JanGVH, 2022-11-23:
I scale the camera control UI to a certain spot on my screen.
I would like it if it came back there when reopening the plugin.

glikely, 2022-11-28:
Can you provide more details please? What platform are you running on. Save & restore of window position works fine for me on MacOS

closed 2022-11-28 by glikely

playaspec, 2023-01-27:
I encountered a similar issue with two other plugins. I'd set them up, even locking the docks, quit, reopen OBS, and they'd be reset. I was about to open an issue with each plugin's author, when I read a comment on a closed issue suggesting to set up plugin docks as desired, take OBS out of full screen, then quit and reload.

Wouldn't ya know, it fixed the problem! Some weird bug with OBS itself not remembering dock positions while maximized to full screen. Once the preference was written, OBS started maximized with all docks where they should be.

Anyone running across this, let us know here if it fixed similar.

glikely, 2025-05-01:
Sounds like this is resolved. Closing the issue.