glikely/obs-ptz #121: choosing another preset

JanGVH, 2022-11-23:
When choosing another preset I have to double-click.
this is difficult with a quick scene change.
can that be a single click?

rasamassen, 2022-12-05:
Agreed. A single-click option or default would be so much better. I don't have to double-click the camera movement controls. Why do I have to double-click the presets?

markusd1984, 2023-05-06:
This may work as a safety mechanism so you don't accidentally change the camera preset/view which would be quite noticeable during a live stream, thus probably worthwhile the extra step and keeping it. 

Perhaps we could have a smaller button next to the preset for instant switching or a setting to change from double click to single click?

glikely, 2025-05-05:
Yes, the double click is because you would also click on a preset to edit it without activating it. I don’t want to break that workflow. However, I like the idea of adding an activate button on the preset row

glikely, 2026-08-16:
Now implemented in main. Will be part of v0.20.x major release

closed 2026-08-16 by glikely