glikely/obs-ptz #123: Camera Label above Presets

rasamassen, 2022-12-05:
Especially when using "pre" or "live" to automatically switch which camera is being controlled (so that you're rarely manually clicking on any camera options), it would be really helpful if, above the presets, the name of the currently selected/controlled camera were displayed. It's easy for a user to forget. Even without pre or live, if it's been a hot second since the user last moved a camera, they might accidentally assume a different camera is selected than the one they want.

kelk10, 2022-12-28:
Like this idea. Currently I used preset 16 and just named it the Camera name to give another visual indication of which camera I am on.

glikely, 2026-10-19:
I'm closing this issue because there is a larger dock layout refactor that needs to be done which makes this specific request obsolete.

closed 2026-10-19 by glikely