glikely/obs-ptz #125: Saved Patterns / Cruises / Tours

rasamassen, 2022-12-08:
It would be great, in addition to regular presets, if we had the ability to create a tour where the program directs the camera to go to multiple points, one after the other, in a set pattern and possibly on a loop. Right now when we want to pan the audience for example, that all has to be done manually, and honestly should be done with some zooming because of the angles of some of our cameras. But that's really hard to do with remote control. It would be great if we could simply save a tour for our users to click and run.

normen, 2023-02-24:
Suggestion: Use the excellent "Advanced Scene Switcher" plugin together with this. Automate calling the presets via keyboard shortcuts.

I am using this to automate three cameras with 4 presets each that get called at random, giving me a total of 12 camera angles in the resulting video.

glikely, 2025-05-07:
I'm closing this one as there are no plans to implement a patterns feature at this time.

closed 2025-05-07 by glikely