glikely/obs-ptz #126: Camera zoom drift after starting OBS

christianhva, 2022-12-09:
This happens right after OBS (28) starts: All my cameras (Birddog P200) starts to slowly zoom before any preset is called. This results in all my cameras beeing zoomed all the way in before i manage to get the production started. 

If i call a camera and its preset, the zoom stops on that camera. 

I had no issue before the obs-ptz plugin was installed. 

Thank you for reading!

christianhva, 2022-12-09:
obs-ptz version 0.13.0

glikely, 2023-01-08:
I've seen this happen recently too, but I haven't been able to reliably reproduce it. I'll keep working on it.

Tscherny-git, 2023-01-10:
I'm not shure, whether it's connected to what I experienced:
I can controll my PTZ-Camera via Pelco-D and Pelco-P, everything but the zoom-buttons works fine. As soon as I press one of the zoom-buttons (doesn't matter if it's zoom-in or zoom-out), it starts to zoom-in to the max. The only way back is to call a preset.

I'm new to the subject, so it could be a mistake in the settings, but as far as I See, the camera settings and the obs-ptz-settings match. (As everything else is working).

Thank you for your amazing work.

bharken, 2023-01-11:
> I'm not shure, whether it's connected to what I experienced: I can controll my PTZ-Camera via Pelco-D and Pelco-P, everything but the zoom-buttons works fine. As soon as I press one of the [...]

This is the same exact issue we are experiencing right now. Zoom in or out will zoom in the camera to the max.

Thanks for this amazing plugin!

glikely, 2023-01-16:
@christianhva , @bharken, @Tscherny-git could you please do the following to help me debug?

1. Go into PTZ settings and turn on "Enable Debug Logging"
2. Restart OBS
3. Don't touch any of the PTZ controls. I just want to see the startup behaviour.
4. Close OBS
5. Attach the OBS log file to this issue

Thanks,
g.

glikely, 2023-01-16:
Also, please try the v0.13.1-rc1 test release to see if it fixes the issue:

https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc1

Tscherny-git, 2023-01-17:
> @christianhva , @bharken, @Tscherny-git could you please do the following to help me debug?
>
>     1. Go into PTZ settings and turn on "Enable Debug Logging"
>
>     2. Restart OBS
>
>     3. Don't touch [...]

[2023-01-17 16-02-24.txt](https://github.com/glikely/obs-ptz/files/10436414/2023-01-17.16-02-24.txt)

Tscherny-git, 2023-01-17:
> Also, please try the v0.13.1-rc1 test release to see if it fixes the issue:
> 
> https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc1

For me, it didn't fix the issue. It's still zooming in to the max as soon as I press one of the zoom-buttons.
Here is another log-file of the startup after installation of the new release.
[2023-01-17 16-08-48.txt](https://github.com/glikely/obs-ptz/files/10436429/2023-01-17.16-08-48.txt)

glikely, 2023-01-17:
> For me, it didn't fix the issue. It's still zooming in to the max as soon as I press one of the zoom-buttons. Here is another log-file of the startup after installation of the new release. [...]

I think you've got a different issue. Please file a separate issue and attach a log file from OBS, but this time I want you to click the zoom button so the problem occurs before shutting down OBS and attaching the log.

glikely, 2023-01-18:
I think this one is fixed in v0.13.1-rc1. At least I was able to reproduce on my end and I think I've found the bug. Please reopen this bug if you're still seeing the problem with v0.13.1-rc1 or newer

https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc2

commit c904d039 (2023-01-16): Fix camera drift on startup

A number of uninitialized variables caused unintentional camera moves to
be sent to the devices on startup. The uninitialized variables were
interpreted as movement that the camera was supposed to be making.
Initialize all member variables to '0' so this doesn't happen

Fixes: #126
closed 2023-01-18 by glikely