glikely/obs-ptz #128: Windows EXE flagged as malicious

benonymity, 2022-12-15:
The Windows EXE installer is flagged as malicious by Cynet on Virustotal--though I know that it is not, and the checksums are all correct, there must be something that raises the hackles of that particular antivirus. Maybe worth researching?

[image]

glikely, 2023-01-08:
First I've seen this. It possibly is related to the fact that I'm not signing the Windows binaries yet.

jonata, 2023-02-23:
@glikely , [SignPath](http://signpath.io/) provides free certificate [for open source projects](https://signpath.org/), if you are interested.

glikely, 2023-02-24:
I am interested. The problem currently is lack of time. If you could walk me through how to get the Windows build signed then I will add it in.

closed 2025-04-30 by glikely