glikely/obs-ptz #129: Option to disable backchannel?

normen, 2022-12-29:
Hi,

first of all thanks for this plugin, its really nice to be able to control PTZ cams directly from OBS.

Since the latest release there seems to be support for cameras reporting back their status to smoothen the handling. My cameras however are connected via a one-way RS485 connection. As there is no feedback from the cams the plugin now reacts much slower and even drops some commands, presumably because its waiting for the cameras to acknowledge the commands.

So I was wondering if it was possible to add an option to disable the backchannel for serial connections.

Thanks again for the plugin and the attention!

glikely, 2023-01-08:
I think I can fix this one in a better way. They main change is the protocol now uses a state machine that rate limits the number of commands that are sent out. However, if the device doesn't send reply, then the rate limit time is based on the timeout timer. I can change the behaviour to have a polling rate that ensures commands are issued in a timely manner, even when replies are not received from the device.

normen, 2023-01-17:
Sounds good, right now the behavior is quite erratic.

glikely, 2023-01-19:
Please try v0.13.1-rc4 and report back

https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc4

glikely, 2023-01-19:
I think this issue is fixed. Please reopen if it is not

closed 2023-01-19 by glikely

normen, 2023-01-21:
Works like a charm, thank you very much!