glikely/obs-ptz #131: Camera zoom - zooms in to the max

Tscherny-git, 2023-01-18:
I'm controlling my PTZ-Camera via Pelco-D (behavior with Pelco-P is the same).
As soon as I press one of the zoom-buttons (doesn't matter if it's zoom-in or zoom-out), it starts to zoom-in to the max. The only way back is to call a preset.

Here is the Log-File when I startet OBS, zoomed-out [then it zooms out for one moment and after that it zooms in to the max] and closed OBS.
[2023-01-18 18-38-39.txt](https://github.com/glikely/obs-ptz/files/10449119/2023-01-18.18-38-39.txt)

glikely, 2023-01-18:
Thank you for the log. I'm assuming this is a new bug in v0.13.0. Can you please try the same thing with v0.12.2 and attach the log? I'd like to compare the behaviour

closed 2023-01-18 by glikely

Tscherny-git, 2023-01-19:
> Please try v0.13.1-rc2 and report back
> 
> https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc2

Thank you so much, it's working now!
When I press the zoom-out button, at first the camera zooms in and after that it zooms out. So, when I push the zoom-out button only once, in the end the zoom doesn't change. But that's not a big deal, because if you hold the button, it zooms out. This doesn't appear on the zoom-in button.
If you want to check the behaviour, here is the log.

Thanks again for your amazing work.
[2023-01-19 14-06-25.txt](https://github.com/glikely/obs-ptz/files/10456985/2023-01-19.14-06-25.txt)

glikely, 2023-01-19:
As far as I can tell, the plugin is sending the correct commands:

```
14:06:36.973: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:25:00:02:28
14:06:36.973: [obs-ptz] PTZPelco::do_update():234: pan 0.000000, tilt 0.000000, zoom -0.050000, focus 0.000000
14:06:36.973: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:40:00:00:41
14:06:37.082: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:25:00:05:2b
14:06:37.082: [obs-ptz] PTZPelco::do_update():234: pan 0.000000, tilt 0.000000, zoom -0.100000, focus 0.000000
14:06:37.082: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:40:00:00:41
14:06:37.148: [obs-ptz] PTZPelco::do_update():234: pan 0.000000, tilt 0.000000, zoom 0.000000, focus 0.000000
14:06:37.148: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:00:00:00:01
```

That's the sequence for a zoom out. It sends a set speed command `00:25`, then zoom wide `00:40` twice; the second speed command making the speed slightly faster. Finally it sends the stop command `00:00`. I suspect this is just the behaviour of the camera. Have you tried with any other controlers? Or does it behave differently with an older version of obs-ptz?

commit 721916ed (2023-01-18): pelco: Fix uncontrolled zoom

Commit 8158a65b, "Cache movement commands so backend can throttle" broke
PELCO zoom control by not sending the correct stop command. The code was
looking for a never satisfied condition (abs(speed) < 0). It also had
the bug that if zoom and pan/tilt were operating at the same time, but
zoom changes would cancel the pan/tilt movement. Fix this by rewriting
the update function to collect the pan, tilt and zoom movements into a
single message.

Fixes: #131
commit 6567bfab (2023-01-19): pelco: Fix uncontrolled zoom

Commit 8158a65b, "Cache movement commands so backend can throttle" broke
PELCO zoom control by not sending the correct stop command. The code was
looking for a never satisfied condition (abs(speed) < 0). It also had
the bug that if zoom and pan/tilt were operating at the same time, but
zoom changes would cancel the pan/tilt movement. Fix this by rewriting
the update function to collect the pan, tilt and zoom movements into a
single message.

Fixes: #131