glikely/obs-ptz #132: pelco: using Strg or Shift + pan-tilt Camera simultaneously

Tscherny-git, 2023-01-18:
When I hold Strg (or Shift) and click on the buttons to move the Camera to the left/right/up/down, it works.
But when I hold Strg (or Shift) and click one of the buttons that moves the Camera in two directions simultaneusly (up-left/up-right/down-left/down-right), the camera crashes. I can't controll it neither can I get the video-input in OBS anymore. I have to restart the Camera and the PC.

Here is the log, when I started OBS, hold STRG and clicked on right-down and closed OBS.
[it appeared in v0.12.2 as well]
[2023-01-18 19-47-27.txt](https://github.com/glikely/obs-ptz/files/10449619/2023-01-18.19-47-27.txt)

glikely, 2023-01-18:
Please try v0.13.1-rc2 and report back

https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc2

Tscherny-git, 2023-01-19:
> Please try v0.13.1-rc2 and report back
> 
> https://github.com/glikely/obs-ptz/releases/tag/v0.13.1-rc2

Didn't fix the issue. Here is the log (same procedure as mentioned above: start OBS, hold STRG and click on right-down, close OBS)
[2023-01-19 14-14-40.txt](https://github.com/glikely/obs-ptz/files/10457013/2023-01-19.14-14-40.txt)

glikely, 2023-01-19:
The protocol trace shows the controller doing what I would expect. It sends one packet for a fast move (`ff:01:00:12:3f:3f:2d`), followed by an all-stop packet (`ff:01:00:00:00:00:01`). Nothing is received back from the camera, and I can't figure out how that would break things to the point that the PC needs to be restarted. Can you capture the same log, but hit the diagonal button a couple of times?

What happens if you only restart the camera, or only restart the PC after the failure?

Can you try using something like 232analyzer to manually push commands to the camera and see how it behaves?

https://www.commfront.com/pages/pelco-d-protocol-tutorial

> 14:14:54.196: [obs-ptz] PTZPelco::do_update():234: pan 1.000000, tilt -1.000000, zoom 0.000000, focus 0.000000
> 14:14:54.196: [obs-ptz] PTZUARTWrapper::send():94: COM3 --> ff:01:00:12:3f:3f:2d
> 14:14:54.196: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:12:3f:3f:2d
> 14:14:54.488: [obs-ptz] PTZPelco::do_update():234: pan 0.000000, tilt 0.000000, zoom 0.000000, focus 0.000000
> 14:14:54.488: [obs-ptz] PTZUARTWrapper::send():94: COM3 --> ff:01:00:00:00:00:01
> 14:14:54.488: [obs-ptz] PTZPelco::send():119: Pelco D command send: ff:01:00:00:00:00:01

Tscherny-git, 2023-01-24:
This time, it worked to move it diagonally for a few times, with and without pressing STR or Shift. But after a short time, it crashes: no control of the camera (via OSB nor via remote) and no picture is sent to OBS. Log-File is attached. Restarting the camera (the Power-Button on the remote still works) doesn't change anything.
But maybe it's the cameras fault (FoMaKo, FMK12UH). For me it's no problem as I don't use the faster or slower movement. Just wanted to give the feedback for you to check whether it's important for your development.

I will try to manually push commads to the camera as you described and report back.
[2023-01-24 15-48-14.txt](https://github.com/glikely/obs-ptz/files/10491234/2023-01-24.15-48-14.txt)

glikely, 2023-01-24:
I don’t see anything in the log that gives me concern. The controller is transmitting only. There is no response back. If the camera stops responding to the remote also, then that tells me the camera is buggy.

Do you get control back by physically cycling power to the camera, without restarting OBS?

glikely, 2023-01-29:
Closing this issue as it doesn't look like a plugin problem, but rather a camera bug. However, please reopen this issue if you have more data

closed 2023-01-29 by glikely