glikely/obs-ptz #133: obs-pts plugin resetting 'view H & V flip' settings.

chewiefortytwo, 2023-01-20:
I have a new Zowietek camera https://www.amazon.co.uk/dp/B0BL7GPYZZ?ref=ppx_yo2ov_dt_b_product_details&th=1

It is mounted upside down to get the best use of the range of movement. It has been setup through the browser interface to flip both in the horizontal and vertical directions. Presets are setup and saved through the browser.

I have loaded the obs-ptz plugin into obs and configured control-panel and browser-source, and all is well, except the video is reset to the default upright view, so picture is upsides down. I then have to open a browser connection to re-flip these settings.
Is this a hard coded 'feature' or is there a configuration file to save the preferred view.

I also have the older Zowietek camera https://www.amazon.co.uk/gp/product/B08JYFST5X/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1 
This is also mounted upside down, but does not have the problem of resetting the H & V flip settings when OBS is loaded.

I have a work around of opening a browser window and resetting the View options, but a fix would be nice to allow less technical operators a better experience.
Thanks

glikely, 2023-01-20:
How are you connecting your camera to OBS? Can you attach your OBS log file please?

chewiefortytwo, 2023-01-20:
Setup is installed in a local church.  Camera is connected through a PoE 1Gb switch as is the OBS pc.  Plugin is using VASCA-IP.  I'll get logs asap.

glikely, 2023-01-20:
Are you certain the Ptz plug-in is causing the reset? Can you try removing the Ptz configuration for that camera, leaving only the video feed enabled, and see if the problem still happens?

chewiefortytwo, 2023-01-21:
[2023-01-19 14-58-34.txt](https://github.com/glikely/obs-ptz/files/10471938/2023-01-19.14-58-34.txt)
Log file at last close of OBS

chewiefortytwo, 2023-01-21:
Browser window opened and flip & mirror are set to on.
PTZ Windows App was previously installed and used to change camera positions.  App opened and camera is still set with flip & mirror set on and view as required.  PTZ App closed.
OBS opened with camera still selected in PTZ Control plugin and camera view has reverted to default.  Opening browser confirms that  flip & mirror are set to off.  Camera removed from PTZ control and OBS closed from File menu.  Log enclosed
[2023-01-21 10-19-07.txt](https://github.com/glikely/obs-ptz/files/10471956/2023-01-21.10-19-07.txt)
PTZ Windows App opened and flip & mirror are defaulted to off and view is upsides down.
Browser connection opened to camera and flip & mirror set to on and saved.

OBS is opened and browser used to select position presets, view is correct with flip & mirror on.  OBS is closed.  
[2023-01-21 10-27-35.txt](https://github.com/glikely/obs-ptz/files/10471971/2023-01-21.10-27-35.txt)

OBS is opened and camera configured in PTZ Control plugin VISCA/TCP with source of camera1 192.168.1.21:5678  on clicking apply camera is immediately upsides down with flip & mirror reset to off, this is confirmed in the browser view image page settings.  Using browser, flip & mirror are set to on, OBS view reflects this.  OBS is closed and log enclosed.  
[2023-01-21 10-30-22.txt](https://github.com/glikely/obs-ptz/files/10471981/2023-01-21.10-30-22.txt)

OBS opened, camera view is again upsides down with flip & mirror reset to off.  OBS closed, log enclosed.
[2023-01-21 10-37-05.txt](https://github.com/glikely/obs-ptz/files/10471984/2023-01-21.10-37-05.txt)

glikely, 2023-01-22:
> OBS opened, camera view is again upsides down with flip & mirror reset to off. OBS closed, log enclosed. [2023-01-21 10-37-05.txt](https://github.com/glikely/obs-ptz/files/10471984/2023-01-21.10-37-05.txt)

Can you go into the PTZ settings dialog, turn on "Enable Debug Logging", and then capture this last log one more time? Without debug logging turned on the log is missing the traffic from the camera.

chewiefortytwo, 2023-01-22:
Opened OBS and enabled logging, removed camera, source and IP address, applied and exited OBS.
[2023-01-22 14-30-15.txt](https://github.com/glikely/obs-ptz/files/10474664/2023-01-22.14-30-15.txt)

re-booted PC.
Browser opened to check camera saved with flip & mirror set on.
Open OBS and camera viewing correct.  Open PTZ controls, debug still enabled.  Add VISCA-TCP:5678 and select source and add IP address.  Clicking apply flips the screen. Back into browser to set flip & mirror ON and submit.  OBS display flips to correct view.  Exit OBS.
[2023-01-22 14-38-48.txt](https://github.com/glikely/obs-ptz/files/10474665/2023-01-22.14-38-48.txt)

Browser still show flip & mirror as ON.
Start OBS and browser view immediately flips and OBS view inverted,  Close OBS.
[2023-01-22 14-49-10.txt](https://github.com/glikely/obs-ptz/files/10474666/2023-01-22.14-49-10.txt)

glikely, 2023-01-30:
As can be seen in that trace, the plugin has sent 3 commands to the camera on startup:
1) 81:01:00:01:ff  # IF_Clear, sent to address 1
2) 88:30:01:ff   # AddressSet
3) 81:01:00:01:ff  # IF_Clear again, possibly because it timed out the first time

According to the Sony manual I have, neither of those commands should be used on VISCA-over-UDP because they are only for a serial connection. VISCA-over-TCP is not a Sony thing, and the zowietek manual doesn't say anything about whether or not they are valid. The address set command actually makes some sense to leave in to support UART-over-TCP bridges. However, I think the IF_Clear is not necessary, and I suspect it is causing the flip settings to be cleared. I'm going to try removing it and see what happens.

Please give this test build a spin and let me know how it goes. The binaries are listed at the bottom of the page.

https://github.com/glikely/obs-ptz/actions/runs/4049184403

chewiefortytwo, 2023-01-31:
Issue unresolved.  Flip & mirror are on the 'Image' page of the browser 
[image]
Old version was uninstalled and then the new version installed, again as soon as the camera details were applied, the view flipped.
[2023-01-31 10-27-14.txt](https://github.com/glikely/obs-ptz/files/10545556/2023-01-31.10-27-14.txt)
[2023-01-31 10-26-52.txt](https://github.com/glikely/obs-ptz/files/10545557/2023-01-31.10-26-52.txt)
I'm not a programmer, so apologies, but I cannot see any variables for the flip & mirror settings.  I have been looking at the PTZ Optics VISCA commands where it does list commands for flip & mirror.
Again apologies, I really appreciate the work you folk can do, because I cannot.

glikely, 2023-01-31:
Very odd. Unfortunately the second log is missing the debug details. Can you please double check that debug logging is enabled in the PTZ settings and retry with the test build so I can see the traffic?

chewiefortytwo, 2023-02-01:
Sorry, just realised there was a download with your previous request.  It is a .pkg file which I now have, but what do I do with it?

glikely, 2023-02-01:
The .pkg file is for installing on MacOS. You need the windows binary. However, the test build doesn't create the installer. You would need to copy the plugin files from the Windows version into the obs plugins directory. However, that's a bit fiddly, so I've tagged a release for you to test instead.

Please try with this release:
https://github.com/glikely/obs-ptz/releases/tag/v0.14.0-pre2

chewiefortytwo, 2023-02-02:
Fantastic, the test release has fixed the 'flipping issue', however the 'pan controls' have also flipped.  Pan up moves the camera down etc.  This didn't happen when I was resetting the camera from the browser.  This is not an immediate issue as I will mainly use the presets to set the camera positions.  Thanks for your prompt responses.  Thought I had sent this from the site with the last log, but clearly I didn't hit the 'comment' button.  Please get back to me if any further logs are required.

normen, 2023-02-02:
In 0.14-pre2 the up/down is also flipped for me (Visca cameras via 485 serial, no backchannel), I guess it has nothing to do with the core issue.

glikely, 2023-02-02:
Glad it is working now. That confirms the camera is resetting the image flip when it receives the IF_CLEAR command. It shouldn't do that, but it is easy enough to drop it for all cameras.

The flipped direction controls that you have now is just weird. None of that logic has changed. With it up and running, can you try switching image flip on and off from the web interface and see if the behaviour changes?

glikely, 2023-02-02:
> In 0.14-pre2 the up/down is also flipped for me (Visca cameras via 485 serial, no backchannel), I guess it has nothing to do with the core issue.

I’ve flubbed something then. What is the last version that worked properly for you?

chewiefortytwo, 2023-02-02:
I'll check the web interface and get back to you, but I had previously installed the 'PTZOptics PTZ Controller 1.4.1 Windows App' as it had check boxes to flip & mirror the image, (their OBS plugins were too big and clunky to use) and I did open that to check the pan controls, which still operated as expected, while at the same time your controls were inverted.

glikely, 2023-02-03:
Yup, I had an unrelated patch in the tree that broke tilt direction on VISCA. I've fixed it in mainline and it will be in the next major release. However, since I'm not quite ready to make the v0.14.0 release yet, I've also tagged a v0.13 point release specifically to fix the H&V flip problem. You can find it here:

https://github.com/glikely/obs-ptz/releases/tag/v0.13.2

closed 2023-02-03 by glikely

normen, 2023-02-03:
> I’ve flubbed something then. What is the last version that worked properly for you?

pre-1 seems to work normal. But it doesn't yet have the nice "save preset" mappings so I just flipped my keyboard mapping for now 😄

commit 41087354 (2023-01-31): visca: Only send IF_CLEAR on UART connections

IF_CLEAR may be causing some cameras to reset settings that shouldn't be
reset. There doesn't appear to be a need to send the IF_CLEAR command on
startup, so remove it from the UDP and TCP implementations.

Fixes: #133 (hopefully)
commit 15e14302 (2023-02-03): visca: Only send IF_CLEAR on UART connections

IF_CLEAR may be causing some cameras to reset settings that shouldn't be
reset. There doesn't appear to be a need to send the IF_CLEAR command on
startup, so remove it from the UDP and TCP implementations.

Fixes: #133 (hopefully)

commit 4108735 upstream
commit 90bbf57d (2023-02-03): visca: Only send IF_CLEAR on UART connections

IF_CLEAR may be causing some cameras to reset settings that shouldn't be
reset. There doesn't appear to be a need to send the IF_CLEAR command on
startup, so remove it from the UDP and TCP implementations.

Fixes: #133 (hopefully)

commit 4108735 upstream