glikely/obs-ptz #135: Flatpak support

lal12, 2023-02-07:
It would be nice if this could be installed via flatpak.

The following config works for me to build it:

```
id: com.obsproject.Studio.Plugin.ptz
name: OBS PTZ Plugin
branch: stable
runtime: com.obsproject.Studio
runtime-version: stable
sdk: org.kde.Sdk//6.4
build-extension: true
separate-locales: false
appstream-compose: false
modules:
- name: obs-ptz
  buildsystem: cmake-ninja
  no-make-install: true
  post-install:
  - install -Dm644 obs-ptz.so ${FLATPAK_DEST}/lib/obs-plugins/obs-ptz.so
  - mkdir -p ${FLATPAK_DEST}/share/obs/obs-plugins/obs-ptz
  - cp -r data/locale ${FLATPAK_DEST}/share/obs/obs-plugins/obs-ptz
  sources:
  - type: git
    url: https://github.com/glikely/obs-ptz.git
    tag: v0.13.2
```

closed 2025-05-01 by glikely

connorbrinton, 2026-01-11:
As of January 11th, 2026, the above manifest does not work with the latest version of obs-ptz due to an error about not working with in-source builds. I was able to modify the above manifest to get it working with the following changes:
```yml
id: com.obsproject.Studio.Plugin.ptz
branch: stable
runtime: com.obsproject.Studio
runtime-version: stable
sdk: org.kde.Sdk//6.8
build-extension: true
build-options:
  prefix: /app/plugins/ptz
separate-locales: false
appstream-compose: false
modules:
  - name: obs-ptz
    buildsystem: cmake-ninja
    builddir: true
    sources:
      - type: git
        url: https://github.com/glikely/obs-ptz.git
        tag: v0.18.2
```

Hopefully that's helpful to someone! 🙂

Sokoloft, 2026-09-01:
Hi, 

Could you please post instructions in the readme on how to install this via flatpak? I'm using Fedora 44. 

I've gone through a hellacious rabbit hole since I switched to the RPM fusion version of OBS to use this plugin. I prefer it not sandboxed. However, it breaks dependencies I need for OBS/DistroAV's NDi plugin and I cant just install ffmpeg 7 to fix it.

Thank you.

Edit: I've since ran the above manifest in chatgpt and I will try what it told me on the target system tomorrow. I'll change the tag based on what release I need per obs version etc.

Sokoloft, 2026-09-02:
Okay it worked. I edited the manifest to use tag `19.1` and I had to use the latest KDE SDK version of `6.11`. The following:

```yml

id: com.obsproject.Studio.Plugin.ptz
branch: stable
runtime: com.obsproject.Studio
runtime-version: stable
sdk: org.kde.Sdk//6.11
build-extension: true
build-options:
  prefix: /app/plugins/ptz
separate-locales: false
appstream-compose: false
modules:
  - name: obs-ptz
    buildsystem: cmake-ninja
    builddir: true
    sources:
      - type: git
        url: https://github.com/glikely/obs-ptz.git
        tag: v0.19.1

```

I needed to install `flatpak-builder` via dnf.
```bash
sudo dnf install flatpak-builder
```

Then I also needed the current KDE SDK.
```bash
flatpak install flathub org.kde.Sdk//6.11
```

 Copy the above manifest file and save it as:
```bash
com.obsproject.Studio.Plugin.ptz.yml
```

Install and build with:
```bash
flatpak-builder --user --install --force-clean build com.obsproject.Studio.Plugin.ptz.yml
```

Worked great. If maintainer can maintain a up to date yml with instructions in the readme that'd be great. 
If not I suppose others will come across this like I did.

Thank you.