glikely/obs-ptz #138: Cannot build the plugin in Fedora 37

notthatadam, 2023-02-12:
I did a git clone on the current version, and it goes ok until I run make.

$ make
[  3%] Automatic MOC and UIC for target obs-ptz
[  3%] Built target obs-ptz_autogen
[  7%] Building CXX object CMakeFiles/obs-ptz.dir/obs-ptz_autogen/mocs_compilation.cpp.o
In file included from /home/adam/Documents/obs-ptz/build/obs-ptz_autogen/UVLADIE3JM/../../../src/ptz-visca-tcp.hpp:11,
                 from /home/adam/Documents/obs-ptz/build/obs-ptz_autogen/UVLADIE3JM/moc_ptz-visca-tcp.cpp:10,
                 from /home/adam/Documents/obs-ptz/build/obs-ptz_autogen/mocs_compilation.cpp:12:
/home/adam/Documents/obs-ptz/build/obs-ptz_autogen/UVLADIE3JM/../../../src/ptz-visca.hpp:37:14: error: ‘optional’ in namespace ‘std’ does not name a template type
   37 |         std::optional<PTZCmd> active_cmd[8];
      |              ^~~~~~~~
/home/adam/Documents/obs-ptz/build/obs-ptz_autogen/UVLADIE3JM/../../../src/ptz-visca.hpp:11:1: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
   10 | #include <QTimer>
  +++ |+#include <optional>
   11 | #include "protocol-helpers.hpp"
make[2]: *** [CMakeFiles/obs-ptz.dir/build.make:100: CMakeFiles/obs-ptz.dir/obs-ptz_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/obs-ptz.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

glikely, 2023-02-12:
Try adding `#include <optional>` to the top of ptz-visca.hpp. Does that fix the build?

notthatadam, 2023-02-12:
Thanks for the quick reply.  I made it farther, but still got an error.

$ make
[  3%] Automatic MOC and UIC for target obs-ptz
[  3%] Built target obs-ptz_autogen
[  7%] Building CXX object CMakeFiles/obs-ptz.dir/obs-ptz_autogen/mocs_compilation.cpp.o
[ 11%] Building C object CMakeFiles/obs-ptz.dir/src/ptz.c.o
[ 15%] Building CXX object CMakeFiles/obs-ptz.dir/src/ptz-controls.cpp.o
[ 19%] Building CXX object CMakeFiles/obs-ptz.dir/src/ptz-device.cpp.o
In file included from /home/adam/Documents/obs-ptz/src/ptz-device.cpp:9:
/home/adam/Documents/obs-ptz/src/ptz-device.hpp:94:23: error: field ‘stale_settings’ has incomplete type ‘QSet<QString>’
   94 |         QSet<QString> stale_settings;
      |                       ^~~~~~~~~~~~~~
In file included from /usr/include/qt5/QtCore/qglobal.h:1301,
                 from /usr/include/qt5/QtGui/qtguiglobal.h:43,
                 from /usr/include/qt5/QtWidgets/qtwidgetsglobal.h:43,
                 from /usr/include/qt5/QtWidgets/qapplication.h:43,
                 from /usr/include/qt5/QtWidgets/QApplication:1,
                 from /home/adam/Documents/obs-ptz/src/imported/qt-wrappers.hpp:20,
                 from /home/adam/Documents/obs-ptz/src/ptz-device.hpp:10:
/usr/include/qt5/QtCore/qtypeinfo.h:216:1: note: declaration of ‘class QSet<QString>’
  216 | Q_DECLARE_MOVABLE_CONTAINER(QSet);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/obs-ptz.dir/build.make:142: CMakeFiles/obs-ptz.dir/src/ptz-device.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/obs-ptz.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

glikely, 2023-02-13:
Try adding #include <QSet> to the top of ptz-device.hpp

What version of OBS studio are you building against? Did you rebuild obs-studio from source, or are you using the packaged version?

I can see that obs-ptz is trying to build against Qt5 instead of Qt6 which suggests you're using an older version of obs-studio (v27 or earlier). The plugin should build against the earlier version, but I've not tested it and that is probably why you're running into these little build problems.

notthatadam, 2023-02-13:
OK, things are starting to make a little more sense.  I didn't realize I needed Qt6 for this.  I have obs-studio 28.1 from rpmfusion.

$ rpm -qa | grep obs
obs-studio-libs-28.1.2-3.fc37.x86_64
obs-studio-28.1.2-3.fc37.x86_64
obs-studio-devel-28.1.2-3.fc37.x86_64

I have a list of Qt5 packages that I needed to install to build the older version, but I'm having trouble finding some corresponding packages for Qt6.  I'm pasting a list of the Qt6 packages that I currently have installed.  Can you tell me what I'm missing there?   As always, thank you for you time.

$ rpm -qa | grep qt6
qt6-qtbase-common-6.4.2-4.fc37.noarch
qt6-qtbase-6.4.2-4.fc37.x86_64
qt6-qtserialport-6.4.2-1.fc37.x86_64
qt6-rpm-macros-6.4.2-1.fc37.noarch
qt6-qtbase-mysql-6.4.2-4.fc37.x86_64
qt6-qtbase-odbc-6.4.2-4.fc37.x86_64
qt6-qtbase-gui-6.4.2-4.fc37.x86_64
qt6-qtsvg-6.4.2-1.fc37.x86_64
qt6-qtbase-postgresql-6.4.2-4.fc37.x86_64
qt6-qtbase-devel-6.4.2-4.fc37.x86_64
qt6-qtsvg-devel-6.4.2-1.fc37.x86_64
qt6-qtserialport-devel-6.4.2-1.fc37.x86_64

notthatadam, 2023-02-14:
That did the trick.  I was able to run make successfully.  Thank you so much.

notthatadam, 2023-02-14:
New issue now, when I drop the plugin into place and try and start OBS, OBS crashes.  The following part is all I see in the log about it:
```
info: [obs-ptz] plugin loaded successfully (version 0.14.1)
QWidget: Must construct a QApplication before a QWidget
Aborted (core dumped)
```
The full log is attached.
[obs-ptz_error.log](https://github.com/glikely/obs-ptz/files/10728544/obs-ptz_error.log)

glikely, 2023-02-14:
I need more information about your build environment in order to debug this. I see you're running OBS v28 built against Qt5 instead of Qt6 which is what the released binaries are built against. Are you building it from source?

Can you provide the exact steps you used for building both obs-studio and obs-ptz? Then I can try to reproduce it on my system.

glikely, 2023-02-14:
Nevermind, I didn't read your earlier comment carefully enough. I think I've got the information I need. I'll try to reproduce using the rpmfusion packages

glikely, 2023-02-14:
I was able to reproduce on a VM here. The RPMFusion obs-studio package is building against Qt5, but obs-ptz links to Qt6 by default. You need to link against the same version to get it to build. I've just pushed out a fix with instructions for changing the Qt version, but I'll copy them here too:

```
git clone https://github.com/glikely/obs-ptz
mkdir obs-ptz/build
cd obs-ptz/build
cmake -DQT_VERSION=5 ..
make
```

closed 2023-02-14 by glikely

notthatadam, 2023-02-14:
It all worked.  I was able to build the plugin, put it in place, and I see it in OBS.  Thank you so much for all of the effort, and for an amazingly useful plugin.

commit fad112bf (2023-02-14): Fix build issues on Fedora 37

A number of problems were exposed when building on Fedora 37 using
RPMFusion packaged obs-studio. Some implicitly included headers needed
to be added. The touch control needed to use the older ->pos() method
for compatibility, and the byte |= operator needed to be modified.

Also update the documentation to cover building against Qt5.

Fixes: #138, Cannot build the plugin in Fedora 37