glikely/obs-ptz #139: Joystick - no zoom control

BenJamesAndo, 2023-02-17:
Running obs-ptz-v0.15.1-joytest1-windows-x64 with Thrustmaster 16000M joystick.
0.15 I could control a few things, now with 0.15.1 I can use the joystick to control pan/tilt. Still no zoom control though.

Attached is the OBS log.
I started out by panning, then tilting. Then twisted the joystick. After that turned the thurst up and down. Then I pressed up, down, left right on the little joystick thing on the top of the main joystick handle. After that I proceeded to press all the buttons on the joystick.

[2023-02-17 22-31-24 PTZ Joystick.txt](https://github.com/glikely/obs-ptz/files/10766717/2023-02-17.22-31-24.PTZ.Joystick.txt)

glikely, 2023-02-18:
Thanks for sending the log file. I can see from the output that the plug-in is receiving input from the joystick, but none of those axes are mapped to controlling zoom.

I’ll rename this issue to be a request for remapping the joystick control.

BenJamesAndo, 2023-02-18:
Great thanks!
Additionally, it would be nice if even more control options were added at some point that could be performed with a joystick. Such as gain up/down, focus in/out & auto-focus, in/decrease zoom/pan/tilt speed. Essentially turning any old joystick into a fully capable PTZ keyboard equivalent.

glikely, 2023-02-24:
I do intend to add remapping of buttons and axes on the joystick to any of the control functions. I've even been looking and wiring the buttons up to the OBS hotkey functions so that they can do anything in the OBS UI.

commit b4a8807c (2025-05-19): joystick: make joystick buttons and axes configurable

Add joystick axis and button mapping to the settings dialog. The mapping
enables a small number of initial actions that can be mapped as the user
likes, including pan/tilt/zoom for the axes, and selecting previous/next
camera or preset, and recalling presets for the buttons. More actions
will get added in the future.

Fixes: #139
commit 8d5bdce9 (2025-05-19): joystick: make joystick buttons and axes configurable

Add joystick axis and button mapping to the settings dialog. The mapping
enables a small number of initial actions that can be mapped as the user
likes, including pan/tilt/zoom for the axes, and selecting previous/next
camera or preset, and recalling presets for the buttons. More actions
will get added in the future.

Fixes: #139
commit 267e8aa7 (2025-05-19): joystick: make joystick buttons and axes configurable

Add joystick axis and button mapping to the settings dialog. The mapping
enables a small number of initial actions that can be mapped as the user
likes, including pan/tilt/zoom for the axes, and selecting previous/next
camera or preset, and recalling presets for the buttons. More actions
will get added in the future.

Fixes: #139