glikely/obs-ptz #142: VISCA focus far/near not working

normen, 2023-02-24:
Hi,

I am using 4 cameras with a unidirectional serial RS-485 VISCA connection to control my cameras which works fine.

However, while switching auto focus / manual focus seems to work fine for my cameras the actual near/far focus commands don't seem to do anything. Using my hardware camera controller I *can* control the focus of my cameras though.

This is the byte sequence that my controller emits when changing the focus near/far:
```
0x81 0x01 0x04 0x08 0x02 0xFF
0x81 0x01 0x04 0x08 0x03 0xFF
```

I didn't check what OBS-PTZ emits, just thought I include this info.

Thanks for this plugin!

glikely, 2023-02-24:
The plug-in will output all the VISCA traffic to the OBS log if you turn on debug logging in the settings dialog. However, just from what you’ve sent I suspect your camera doesn’t like the speed control variant of the focus commands that the plug-in emits.

Please enable debug logging and attach the log to this issue

normen, 2023-02-24:
Yes, OBS-PTZ seems to output these instead:
```
83:01:04:08:27:ff
83:01:04:08:37:ff
83:01:04:08:00:ff
```
The options allow me to cap the values for pan and zoom, but not for focus.. 🙂

normen, 2023-02-24:
Looking into the code a bit the VISCA implementation has the same commands my camera uses here: https://github.com/glikely/obs-ptz/blob/0b5b832b9e234cc3babb465a9e28848be89cbe75/src/ptz-visca.cpp#L287
However whats used seems to be the speed variable version a bit below.

glikely, 2023-02-24:
That's correct. I populated the driver with as many commands as I could find in the Sony SRG-120DH manual, but then only used some of the commands in the initial implementation. What is probably needed here is a set of capability bits that tracks which features are available on a given camera.

normen, 2023-02-24:
Yeah, admittedly the VISCA implementation in these cameras leaves a lot to be desired. But my guess is that this is not uncommon especially with cheap / prosumer devices so what you say sounds like a good idea. Thanks for considering!

glikely, 2023-03-01:
I'm going to rename this issue to be a feature request for capability bits so that unsupported messages can be avoided.

markusd1984, 2023-04-16:
Maybe an alternative could be to enable/disable certain buttons/AF buttons via settings?

I'm very interested in custom commands buttons to perform additional tasks, i.e. most important for us is turning [auto tracking on/off for ePTZ camera](https://github.com/glikely/obs-ptz/issues/155).

glikely, 2026-10-02:
Fixed in mainline

closed 2026-10-02 by glikely