glikely/obs-ptz #144: Not Connecting to Camera?

bassguy95, 2023-02-27:
Using the latest update of the plugin with OBS 28.1.1 - I add my cameras and try to control them but nothing happens...

I know the IP/port settings are correct as they work fine with Companion, I'm just hoping to get the camera control actually inside OBS,

Cameras being used are DataVideo PTC-150.

bassguy95, 2023-02-27:
> Try enabling debug in the settings and open the OBS log to see if theres errors.
`10:25:38.612: [obs-ptz] VISCA_over_TCP PTZ Device 2 connected
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device 2] Stale prop list: {zoom_pos,tilt_pos,pan_pos,color_hue,iris_pos,gain_pos,wb_mode,dzoom_pos,vendor_id,camera_id,defog_mode,focus_pos,power_on}
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device 2] --> 81:09:7e:7e:00:ff
10:25:38.612: [obs-ptz] VISCA_over_TCP PTZ Device connected
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device] Stale prop list: {zoom_pos,tilt_pos,pan_pos,color_hue,iris_pos,gain_pos,wb_mode,dzoom_pos,vendor_id,camera_id,defog_mode,focus_pos,power_on}
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device] --> 81:09:7e:7e:00:ff
10:25:38.612: [obs-ptz] VISCA_over_TCP PTZ Device 1 connected
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device 1] Stale prop list: {zoom_pos,tilt_pos,pan_pos,color_hue,iris_pos,gain_pos,wb_mode,dzoom_pos,vendor_id,camera_id,defog_mode,focus_pos,power_on}
10:25:38.612: [obs-ptz] [visca-over-tcp/PTZ Device 1] --> 81:09:7e:7e:00:ff
10:25:40.115: adding 512 milliseconds of audio buffering, total audio buffering is now 512 milliseconds (source: Video Capture Device)
10:25:40.115: 
10:25:41.033: adding 21 milliseconds of audio buffering, total audio buffering is now 533 milliseconds (source: Video Capture Device)
10:25:41.033: 
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device 1] timeout
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device 1] --> 81:09:7e:7e:00:ff
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device] timeout
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device] --> 81:09:7e:7e:00:ff
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device 2] timeout
10:25:41.757: [obs-ptz] [visca-over-tcp/PTZ Device 2] --> 81:09:7e:7e:00:ff
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device 2] timeout
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device 2] --> 81:09:7e:7e:00:ff
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device] timeout
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device] --> 81:09:7e:7e:00:ff
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device 1] timeout
10:25:42.049: [obs-ptz] [visca-over-tcp/PTZ Device 1] --> 81:09:7e:7e:00:ff
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device 1] timeout
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device 1] --> 81:09:7e:7e:00:ff
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device] timeout
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device] --> 81:09:7e:7e:00:ff
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device 2] timeout
10:25:42.147: [obs-ptz] [visca-over-tcp/PTZ Device 2] --> 81:09:7e:7e:00:ff`

is "81:09:7e:7e:00:ff" supposed to be the camera MAC? Because that's not even close to my camera's MACs.

normen, 2023-02-27:
No, thats a VISCA command.

normen, 2023-02-27:
Are you sure your camera uses VISCA over TCP? Theres also VISCA over UDP.

bassguy95, 2023-02-27:
> No, thats a VISCA command.

I admit I’m not very knowledgeable about this side of things.

> Are you sure your camera uses VISCA over TCP? Theres also VISCA over UDP.

Yes, I am sure. I already got this VISCA over TCP working with an app called Companion. That works fine, but it needs another window open or tablet use, which is great, but I also want control in OBS for some cases. 

With Companion it was as easy as setting the IP of the camera and selecting the port I want to use.

glikely, 2023-03-01:
The plugin appears to be connecting to the TCP port on the camera, but it is not getting any responses, and the connection gets closed immediately. Either something isn't configured right, or the plugin isn't speaking quite the right VISCA dialect for the DataVideo camera.

Do you know how to use Wireshark? If you could get a trace of the TCP messages between Companion and the camera then it would help debug. Also, please attach a full copy of your OBS log file so that I can see the initialisation messages.

I like the suggestion of having visible connection status in the plugin. Please feel free to file a separate issue requesting that feature so that I don't lose track of it.

bassguy95, 2023-03-01:
> Do you know how to use Wireshark? If you could get a trace of the TCP messages between Companion and the camera then it would help debug. Also, please attach a full copy of your OBS log file so that I can see the initialisation messages.

I'm working within a public school district and I don't think our network admin will take to kindly to me sniffing around with Wireshark, but I can ask.

If it helps at all, [Companion](https://github.com/bitfocus/companion) is another open source project designed to allow tablet emulation of the streamdeck as well as some interconnectivity options. 

Incidentally I also found [this plugin](https://github.com/bitfocus/companion-module-datavideo-visca/blob/master/HELP.md) while searching for the official DataVideo documentation as I know I've seen it before. 

I will update when I get a response from my network admin about allowing the use of Wireshark

bassguy95, 2023-03-01:
My apologies!

[2023-03-01 09-16-38.txt](https://github.com/glikely/obs-ptz/files/10861910/2023-03-01.09-16-38.txt)

 This ought to contain what you need down at the bottom! Also super appreciate the help on this!

bassguy95, 2023-03-01:
Thanks for the response! I was about to send over a copy of their actual implementation:

[DataVideo PTZ VISCA Commands Rev 2.3](https://www.datavideo.com/global/file/download?id=4099)

[PTC Series VISCA Commands](https://www.datavideo.com/global/file/download?id=4137)

Hopefully that helps, I wish I could be of more help as my coding skills are about the same level of a 5th grader these days haha.

glikely, 2026-10-02:
I have an untested implementation of the protocol here: https://github.com/glikely/obs-ptz/tree/visca-datavideo-dvip

I need someone who actually has the hardware to test it before I merge it into mainline

closed 2026-10-25 by glikely