glikely/obs-ptz #145: Feature Suggestion: Add Connection Status Indicator

bassguy95, 2023-03-01:
Something as simple as a little red dot for no connection, green for connection, and perhaps yellow for error? 

This would be super useful for quickly diagnosing problems with cameras.

markusd1984, 2023-04-16:
Wouldn't most if not all people though have the camera as a source displayed in OBS and thus seeing the feed means the camera is working vs seeing no input?

bassguy95, 2023-04-16:
> Wouldn't most if not all people though have the camera as a source displayed in OBS and thus seeing the feed means the camera is working vs seeing no input?

Not if they are using a hardware mixer and ONLY using OBS for graphics and streaming. 

IE, some running a Blackmagic with 3 cameras, using the program output to feed a capture card on a computer. 

In this setup, there is no way to make each camera a separate source, as they're all the same source. I can't imagine this is an uncommon setup, as directly capturing multiple streams from a USB dongle is CPU intensive, as is capturing multiple streams from a PCI capture card. 

Using a hardware mixer greatly reduces the load on the machine that's actually doing the streaming. 

Same goes for audio inputs when you're mixing over 30 microphones placed around a school stadium. Way more practical to use a hardware mixer and feed a single stereo feed into OBS.