glikely/obs-ptz #149: Only Preset 1 Will Recall

cigdangle, 2023-03-26:
Set up my camera using ONVIF.  Everything works well, except recalling presets other than Preset 1.  I can see (using ONVIF Device Manager) that the presets are created (PRESET_0, PRESET_1, PRESET_2, etc.), but only double-clicking "Preset 1" in the PTZ Controls dock will actually recall anything.  The other presets created using the plugin (PRESET_1, PRESET_2, etc.) will recall if I use ONVIF Device Manager to GoTo them.

- Amcrest IP3M-941B
- OBS 29.0.2 (64-bit)
- VLC 3.0.18 (64-bit)
- OBS PTA Controls Plugin 0.16.0-test1

glikely, 2023-03-26:
Thanks for the the feedback. ONVIF support is not yet complete. Preset support in particular isn’t working correctly yet, but it will be fixed before I make an official (non-test) release with ONVIF enabled.

closed 2023-03-26 by glikely

cigdangle, 2023-03-26:
Something I figured out - if you exit OBS and re-open it, the presets all seem to work!