glikely/obs-ptz #152: macOS arm64, universal, and x86_64 packages place plugin in deprecated plugin folder location.

revpjbbq, 2023-04-01:
On macOS Ventura 13.2.1 and 13.3 (only versions available to test) and an Apple Silicon M1 MacBook Air with freshly installed macOS running as a standard (non-admin) user, the provided release packages for obs-ptz v0.15.4 and v0.16.0-test2 pre-release all use the path
```
system root/Library/Application Support/obs-studio/plugins
```  
to place the obs-ptz.plugin file.

When running OBS nothing is added to the Docks menu after installing the package and running the OBS Apple Silicon 29.0.2 release.

This plugin folder location is deprecated and OBS now looks in the logged in user's Library for plugins. I dragged the plugins folder that contained the obs-ptz.plugin file into
```
/Users/logged in user/Library/Application Support/obs-studio/
```
as that folder did not yet exist.

Launching OBS now successfully loads the obs-ptz plugin and I now have PTZ Controls as expected.

glikely, 2023-04-06:
I've not been able to reproduce in my environment, but I'm running and testing from an administrator account. This plugin uses the install location as set up in the plugin template. If this is indeed an issue then it should also be reported there. Any fix I do here I'll want to get committed to the template project first.

https://github.com/obsproject/obs-plugintemplate/issues

Can you point me to the documentation or commit that says the `Application Support/obs-studio/plugins directory` is deprecated?

glikely, 2025-04-19:
The build system has been updated to the latest plugin template. Hopefully this is now correct and will be fixed in the next release.

closed 2025-04-19 by glikely