glikely/obs-ptz #156: PTZ Plugin won't show up in Docks menu

martialgallorini, 2023-04-19:
Hi,

Using OBS 29.0.2 and latest PTZ Plugin (also tried with downgraded versions of both : either i don't understand or i have an issue because no plugin is shown under Docks menu (and i can't find it anywhere else...) so i am unable to use it.

Re-installed multiple times.

Tried under Macos and Linux.

thanks

dvsapp, 2023-04-22:
Same here OBS 29.0.2.1 with latest PTZ plugin. Running Siduction "Masters of War". No plug showing up in the menu. It appears the files are installed into the /usr/lib/..... directories as they should, but OBS apparently can't find it.

martialgallorini, 2023-04-24:
Update : just had to install an Arch based distribution for a project and had a shot on OBS PTZ Control just in case... and it seems to work ! 

Tried 4 different other distributions (Debian based) and none is working with PTZ control.

Something wrong on those platforms ?

Thanks

dragoneo71, 2023-08-11:
got the same troubles but on windows using obs 25.0.4 64bit and latest plugin 16.0, there's nothing in the dock about ptz

glikely, 2025-04-20:
I'm closing this issues as a duplicate. #165 tracks the same issue.

By way of update, I have got the plugin building again against the latest OBS studio, but I had to disable serial port support in the process due to QSerialPort not being included in the obs-deps prebuilt package. Getting it re-enabled won't be easy unfortunately.

closed 2025-04-20 by glikely